FreeCAD/FreeCAD #14721: Grounded joint icon cannot be hidden

furgo16, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Grounded Joint icon (red lock) is always visible in an assembly and cannot be hidden

[image]

There are some inconveniences to this:

1. The icon permanently obscures part of the assembly
2. The icon has a fixed size: on big assemblies where zooming out is necessary, the size of the icon can obscure an even more significant part of the assembly
3.  There is no way to obtain a screenshot of an assembly that does not feature the lock icon: neither using the <kbd>Tools</kbd> > <kbd>Save image...<kbd> command, nor via [TechDraw_ActiveView](https://wiki.freecad.org/TechDraw_ActiveView), nor via any OS-specific screen capture tool.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37731 (Git) Snap 1052
Build type: Release
Branch: main
Hash: 6623b605282aa5ae61ae3df0adff2f6c857477fa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.5
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-25:
I cannot reproduce this bug. If I toggle the visibility of the grounded joint the lock disappear correctly on my side.

furgo16, 2024-06-25:
I did not know that that was the way to hide the lock. I had searched preferences, turned off the visibility of the grounded component... but it did not occur to me that the grounded joint was set to visible, and that it was the reason for the lock to appear.

I've updated the documentation to mention this => https://wiki.freecad.org/Assembly_ToggleGrounded#Notes

Closing this issue report as not a bug. Thanks.

furgo16, 2024-06-25:
Closing this issue report as not a bug. Thanks.

closed 2024-06-25 by furgo16