FreeCAD/FreeCAD #14722: Undoing a fixed joint results in continuous output of <class 'RuntimeError'> messages

furgo16, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open the attached file with FreeCAD: [SimpleAsy2.zip](https://github.com/user-attachments/files/15859644/SimpleAsy2.zip)
1. Add a fixed joint (Assembly_CreateJointFixed command)
2. When the "Create Joint" Tasks dialog appears, click on the top face of the cube to select it
3. Click on the cylinder body
4. The two components will be assembled => OK
5. Press the undo toolbar button or "Ctrl+Z" on the keyboard
6. The Report view will start continually outputting the `21:32:51  <class 'RuntimeError'>` line

The printing of that error can only then be stopped by clicking the <kbd>Cancel</kbd> button in the Tasks dialog

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37731 (Git) Snap 1052
Build type: Release
Branch: main
Hash: 6623b605282aa5ae61ae3df0adff2f6c857477fa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.5
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-25:
This problem is a widespread one. In many tools using ctrlZ (undo) is causing a lot of issues.

This is because "Undo" is not handled by the current tool but by core. So it is undoing the last opened command (commited or not) and cannot be handled by the task, here for example the task would want to undo the selection of the JCS.

Other places where this causes problems : 
- Sketcher : start drawing a rectangle, CtrlZ : undo last command instead of unsetting the first point.
- Sketcher : Dimension tool : CtrlZ : nothing as 'Ctrl+Z' is caught, but if you click on the Undo manually it seg fault (there's an issue about this).

@wwmayer what do you think about this? Is there a proper way to handle 'Undo' when there's an active task?

furgo16, 2024-06-25:
Related issue from last comment: https://github.com/FreeCAD/FreeCAD/issues/12033

Is this one also related? https://github.com/FreeCAD/FreeCAD/issues/9789

wwmayer, 2024-06-27:
> what do you think about this? Is there a proper way to handle 'Undo' when there's an active task?

I don't know the Assembly code in detail but usually a task dialog has to make sure to open a transaction if not yet done when showing it. In its accept() method it commits the transaction and in its reject() method it aborts it.

If a task dialog needs more control when calling undo/redo then it should connect the signalUndo or signalRedo signals of the App::Document.

Sketcher is a bad example in this context because it works completely different. Here every command opens and commits a new transaction while the dialog is opened. That's why it only has a Close button and not OK/Cancel buttons.

wwmayer, 2024-06-27:
> The Report view will start continually outputting the 21:32:51  <class 'RuntimeError'> line

This alone doesn't trigger the warnings. They start to appear when moving the cursor inside the 3D view.

PaddleStroke, 2024-07-02:
Note the task is in python : https://github.com/FreeCAD/FreeCAD/blob/f82eef7533752a83f2aae3c37477b83c1229c628/src/Mod/Assembly/JointObject.py#L1257
Perhaps this has something to do with the issue?

wwmayer, 2024-07-02:
> is the reject method supposed to be called automatically by the undo?

No. But the base class TaskDialog has the member setAutoCloseOnTransactionChange() to close the dialog automatically when doing undo/redo. If your sub-class needs to do some clean-up work beforehand it can override autoClosedOnTransactionChange()

PaddleStroke, 2024-07-02:
`setAutoCloseOnTransactionChange` is a cpp function, and I don't seem to be able to access it from the python task class

```
class TaskAssemblyCreateJoint(QtCore.QObject):
        self.setAutoCloseOnTransactionChange(True)
```
says setAutoCloseOnTransactionChange does not exist.

There are no python examples of setAutoCloseOnTransactionChange in the source base.

Will setAutoCloseOnTransactionChange(true) call the reject function or only close the dialog? If only close the dialog how do I override autoClosedOnTransactionChange in python?
Thanks!

wwmayer, 2024-07-02:
It only sets a member variable that will be checked by TaskView::slotUndoDocument/TaskView::slotRedoDocument

> says setAutoCloseOnTransactionChange does not exist.

It's not part of your task box class (i.e. TaskAssemblyCreateJoint) but of the task dialog.

Here a simple example:
```
from PySide import QtWidgets
class TaskPanel:
  def __init__(self):
      self.form = QtWidgets.QCalendarWidget()

doc = App.newDocument()
doc.UndoMode = 1
doc.openTransaction("Add object")
doc.addObject("Part::Box", "Box")
doc.recompute()
doc.commitTransaction()

Gui.Control.showDialog(TaskPanel())
dialog = Gui.Control.activeTaskDialog()
dialog.setAutoCloseOnTransactionChange(True)

doc.undo() # closes the dialog
```

PaddleStroke, 2024-07-02:
Thanks for the reply.

Ok so that works to close the dialog. But this does not call the reject of my dialog.

Perhaps by default when setAutoCloseOnTransactionChange is set to true it should call the reject of the task?

If we cannot change this default, then how can I reimplement the behavior?

wwmayer, 2024-07-02:
> If we cannot change this default, then how can I reimplement the behavior?

autoClosedOnTransactionChange() will be exposed to Python so you can call reject() inside there.

wwmayer, 2024-07-02:
https://github.com/FreeCAD/FreeCAD/pull/15116

wwmayer, 2024-07-02:
```
from PySide import QtWidgets
class TaskPanel:
  def __init__(self):
      self.form = QtWidgets.QCalendarWidget()
  
  def reject(self):
      ...
  def autoClosedOnTransactionChange(self):
      self.reject()

doc = App.newDocument()
doc.UndoMode = 1
doc.openTransaction("Add object")
doc.addObject("Part::Box", "Box")
doc.recompute()
doc.commitTransaction()

dialog = Gui.Control.showDialog(TaskPanel())
dialog.setAutoCloseOnTransactionChange(True)
dialog.setDocumentName(doc.Name)

doc.undo() # closes the dialog
```

Notes:
* showDialog() now returns the task dialog wrapper
* In order to make "autoCloseOnTransactionChange" working you must also set the name of the document. In the past it was also closed when doing undo/redo for any other document
* autoClosedOnTransactionChange is a method of the passed TaskPanel, not TaskDialog

> Though it feels like this dialog object is not the same as the TaskPanel() object no?

They are different objects. The TaskDialog provides the infrastructure to handle a TaskPanel, e.g. the OK/Cancel/Close/... buttons, the container widget to embed the TaskPanel or multiple task panels, ...

closed 2024-07-10 by wwmayer