FreeCAD/FreeCAD #14723: Measure Tool

Danielivy, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am attempting to adjust the font size of the measure tool, but I have not seen any changes. I am unsure if I am making an error or if there is an issue with the software. The steps I follow are: I navigate to Settings, then Part Design, and finally to Measure Tool to change the font size. After making the change, I save the settings and restart the program, but nothing changes.

I have low vision, and increasing the font size would significantly help me. I am using a Mac Mini with the latest operating system, and it is an Intel-based computer. Your assistance in resolving this matter would be greatly appreciated.

Thank you.

### Full version info

```shell
[code]
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.20
  * Design456 0.0.1
  * CurvedShapes 1.0.9
  * Curves 0.6.36
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-17:
Can you try it in the weekly dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
There is a new measurement tool.

Danielivy, 2024-06-17:
Ok, the new measurement tool works thank you so much for the help.

closed 2024-06-17 by Danielivy