FreeCAD/FreeCAD #14725: Make spreadsheet "Content" edit field resizable

krushia, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I find myself copying spreadsheet cells to a text editor often due to the field overflowing. Would be nice to have it resizable (to multiple lines).

[image]

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git) AppImage
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a (Disabled)
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * dodo 1.0.1
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools (Disabled)
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-17:
What exactly should be resizable?

krushia, 2024-06-19:
> What exactly should be resizable?

The "Content" field. Sorry, I should have highlighted it.
[image]