FreeCAD/FreeCAD #14726: Addon manager usability issues

nazar-pc, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I looked at addon manager and found multiple issues with it that I have not seen reported yet:
* [ ] Links on pages rendered in Addons Manager (likely from Markdown) are not clickable
* [ ] There is no link to the repository or whatever the source happens to be for the workbench for example to check the author, source code, etc. before installing it
* [ ] Clicking "Back" button to return from addon page to the list doesn't preserve scroll progress, which is annoying if wanting to check a few addons (like assembly addons, which there are a few of)

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-17:
@chennes FYI

maxwxyz, 2024-06-17:
also related:
- https://github.com/FreeCAD/FreeCAD/issues/13535

chennes, 2024-06-17:
Can you please split, @maxwxyz ? Then assign to me. I'm going to try to fix the scroll thing and the links for 1.0. The feature request will have to wait for post 1.0.

chennes, 2024-06-18:
I can't reproduce the links-not-working issue -- they open just fine on my machines (in my system browser). I will have to spool up an Ubuntu VM to do some further testing. The scroll position is fixed by https://github.com/FreeCAD/FreeCAD/pull/14753

maxwxyz, 2024-06-18:
done

nazar-pc, 2024-07-27:
I'm now thinking this is a wider FreeCAD issue. For example none of the links in "About" dialog are working for me either.

nazar-pc, 2024-07-29:
Okay, the reason is that `gio-launch-desktop` is not packaged with the AppImage and I don't have it on my system, so clicking on links prints log message in CLI that it can't find `gio-launch-desktop`.

maxwxyz, 2024-07-29:
@adrianinsaval FYI

chennes, 2024-07-29:
@maxwxyz can you update the title on this one to reflect this new information?