FreeCAD/FreeCAD #14729: 0.22.0dev.37730: Windows color scheme unusable

probonopd, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In bright mode, the menu selection color is black, making it impossible to read the highlighted menu item.

Also, the drop-down next to some icons is invisible.

[image]

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-06-17:
Should be fixed by https://github.com/FreeCAD/FreeCAD/pull/14693#event-13178535038

maxwxyz, 2024-06-24:
@probonopd is it fixed in the latest weekly https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ?

probonopd, 2024-06-26:
Yes. Thank you very much.

closed 2024-06-26 by probonopd