FreeCAD/FreeCAD #14736: [Problem] Sketcher: no way to "escape from" autosnapping a circle with autoconstrain on

MiqCG, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Sketcher, after "Create external geometry" of a circle of a previous Pad, if "Auto constrains" is active, you can create a coincident circle with the external geometry but all following circles will snap too to the same circle, no matter how far you draw them. 

If after the first circle you draw a second circle with another centre, the first circle ends being totally defined and you can draw multiple circles with the same origin as the first one without problem.

See screencast and attached file.

Looks like it doesn't happen with FreeCAD revision 35966

Maybe related to #13228
I have searched in Github FreeCAD with [is:issue is:open label:"WB Sketcher" snap ] with no more relevant results.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git) AppImage
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.22
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/2c1afaa7-a326-4cd8-820f-c5248d8b1d8f

[Snap.FCStd.zip](https://github.com/user-attachments/files/15866023/Snap.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-17:
Can reproduce on Windows too.
@PaddleStroke @FlachyJoe FYI, weird behavior.

maxwxyz, 2024-06-20:
also a line created after this is broken.

chennes, 2024-07-15:
Pinging either @PaddleStroke or @AjinkyaDahale -- do either of you have a suggestion?

chennes, 2024-07-19:
I cannot reproduce with
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38239 (Git)
Build type: Release
Branch: openglAddMissingIncludeToPCH
Hash: 076d770efb19aa7da8bb23d3305ce0d04209e95b
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
```

@maxwxyz or @MiqCG can you please re-test and confirm that you still see the issue with the latest main?

maxwxyz, 2024-07-19:
@chennes still an issue, also on macOS:
[image]

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git)
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41

```

kisolre, 2024-08-04:
The problem is that the suggested and applies tangency constraint overconstraints the sketch but by some reason (probably because of the fixed external geometry) it is not reported as partially redundant but as redundant, further analysis that would have to remove the redundant constraint fail because they recognize it as partially redundant and then you get "solver failed to converge" error because while geometrically correct such combination of constraints (coincidence and internal tangency of two circles) is not arithmetically solvable (probably because of standard solver limiting conditions?). Following added circles are added to already unsolvable sketch and by some reason they are not immediately/properly drawn. If I hide the first Pad while still in sketch edit mode all circled show, but now properly drawn in red to show the unsolvable sketch status. Strangely enough if now I exit sketch edit mode no part of the sketch is visible. I am not sure how that will effect consecutive operations that used the sketch while it was still solvable. Cant think of easy way to test that.
I tried the shown behavior in the clip with another geometry and some constraints already present in the sketch and then the tangency constraint is properly removed as partially redundant.

obelisk79, 2024-08-15:
I was able to reproduce the error state as well.

I did some testing and noticed that what is happening here is that the solver is throwing an error state as noted by kisolre above, there is an issue with the tangency constraint. The solution may be something as simple as merely having the autcostraint logic apply equality instead of tangency in the case when two arcs have coincident centerpoints. @PaddleStroke

AjinkyaDahale, 2024-08-19:
If anyone has quick access to 0.21, could they please tell what was the behaviour in that?

chennes, 2024-08-19:
0.21.2 created the bad constraint, but showed it as bad, and let you continue drawing the same as with any other solver failure.

AjinkyaDahale, 2024-08-19:
Maybe bisect next?

AjinkyaDahale, 2024-08-19:
It appears we can check replicate this by script itself. For example, while this problem is happening, the following set of commands gives us the wrong circle size:
```
>>> geoList = []
>>> geoList.append(Part.Circle(App.Vector(0.000000, 0.000000, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 25.910128))
>>> App.getDocument('Snap').getObject('Sketch001').addGeometry(geoList,False)
>>> del geoList
```

AjinkyaDahale, 2024-08-20:
Very peculiar:
1. If you draw circles off-origin, then add tangent, and then coincident with origin, there is no issue.
2. If you draw the circle on orignin, or off-origin and then add coincident, and then add tangent, this tangent is proclaimed to redundant.

AjinkyaDahale, 2024-08-20:
Could someone other than me follow the following steps lead to the same outcome:
1. Repeat the steps in the video.
2. In console, run `App.getDocument("<name of document>").getObject("<name of sketch>").Geometry`.
3. Recompute the sketch.

Expected outcome: The output of step 2 has the size of the last created circle correct, and on recomputing, that last circle is also drawn at the intended place.

maxwxyz, 2024-08-20:
@AjinkyaDahale it does not work for me, on recompute it says:
[image]

```
OS: macOS 13.6.9
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36908 (Git)
Build type: Release
Branch: main
Hash: eb842aadffa2615037c45101ac2575fc79438850
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42

```
And it just displays  an old version of the sketch:
[image]

AjinkyaDahale, 2024-08-20:
OK I think I found the issue. Need to confirm if this needs to be wrapped back in try-catch, but that's for tomorrow. See related PR.

commit 56b9b485 (2024-08-26): Merge pull request #15958 from AjinkyaDahale/sk-fix-14736

[Sketcher] Attempt to fix #14736
commit a30bbba4 (2024-08-26): [Sketcher] Fix #14736

Recompute even if the addition of new elements and autoconstraints failed
partially.
closed 2024-08-26 by chennes