FreeCAD/FreeCAD #14740: Proposal for usability improvements, from an architect and civil builder.

gianlucafiore, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am going to make a general summary of what I can appreciate in my feelings when using FreeCAD in my daily life.

To greatly speed up the workflow, the following improvements should be implemented.

1. Let the default dynamic input be length, and not ΔX or ΔY, so that it is more agile when entering a measurement for lines, walls, beams, etc. 

[image]

It would even be as useful if it were implemented in the same way as in sketcher mode
[image]

2. Have more icon sizing options, since to work on a notebook with a 15" screen, 16px icons are very small, and 24" icons are very large
[image]

The 3D viewing space is quite reduced, perhaps 20px in my case resolution would be the right size.
[image]

3. Polar tracing of fictitious lines, fictitious extension with distance input to start drawing. It is very useful to avoid having to draw guide lines that fill the work with garbage, and then having to erase them. as I mention in the forum with the following [topic](https://forum.freecad.org/viewtopic.php?t=87850)
[image]

4. Ability to start drawing and create architectural elements within a group/folder/Level. Because by default everything new that is created is created in the master branch of the tree, and then you have to order the objects one by one, with the possibility of breaking hierarchies, or opening the door to errors of inconsistencies and orphanhood. 
[image]

Thank you very much in advance, I await comments or suggestions that may not be known to me.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-17:
1.
This is already possible by changing the "Set focus on Length instead of X coordinate" preference.
See: https://wiki.freecad.org/Draft_Preferences#General

3.
There is a feature called "hold" points.
See: https://wiki.freecad.org/Draft_Snap#Advanced_snapping

kaiwas, 2024-06-17:
Point 4 is very relevant. 
It’s terribly annoying when a scene accumulates many elements and each new one is created at the very bottom and you have to constantly scroll to it and drag it to the right place.
The only thing worse is the loss of data in the fields due to mouse movement)

gianlucafiore, 2024-06-17:
> El punto 4 es muy relevante. Es terriblemente molesto cuando una escena acumula muchos elementos y cada uno nuevo se crea en la parte inferior y tienes que desplazarte constantemente hasta él y arrastrarlo al lugar correcto. Lo único peor es la pérdida de datos en los campos debido al movimiento del ratón)
> 
> > tal vez 20px en mi resolución de caso
> 
> Esto se puede hacer editando el archivo de configuración (user.cfg) en un editor de texto (eso es exactamente lo que hice) ![2024-06-17 22_11_12](https://private-user-images.githubusercontent.com/32273833/340425122-7c080e37-d364-4edc-872b-c8956f2ff924.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2NTM2NzcsIm5iZiI6MTcxODY1MzM3NywicGF0aCI6Ii8zMjI3MzgzMy8zNDA0MjUxMjItN2MwODBlMzctZDM2NC00ZWRjLTg3MmItYzg5NTZmMmZmOTI0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE3VDE5NDI1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWExZGRhNjE1NzkxMTQ0Njk4NmViNjY4NzYxZjVlOTAyYjM1ZTI2OTg1NDFhNzk4ZmI2NWYyMDcxZWVlZGMzMDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.M3hZymNM6nVviRpoLWXSfmR-0y1D5Lfbpwaq-6OF834)

Great thanks for your contribution, where is this file located by default?

yorikvanhavre, 2024-06-24:
> 4. Ability to start drawing and create architectural elements within a group/folder/Level. Because by default everything new that is created is created in the master branch of the tree, and then you [...]

Yes, that's also something we all want. I tried several systems already (using autogrouping, using BuildingParts as "Active Containers...") but none of these were really used and useful.

I'm working now on a revamped BimViews dialog where you can define levels and views as Active. And I plan to make it compatible with the Autogroup system too. Hopefully that'll give a big step forward.

luzpaz, 2024-07-18:
Can we clean this ticket up, perhaps separate it into individual tickets ?

Roy-043, 2025-02-07:
Ad. 4: If you check the "Include groups in layer list" preference you can already activate a group or a group-like object such as a BuildingPart with [Draft AutoGroup](https://wiki.freecad.org/Draft_AutoGroup). Any new object will then be put in the specified group or BuildingPart.

In one form or another all points have been addressed IMO.

kaiwas, 2025-02-07:
> BuildingPart with [Draft AutoGroup](https://wiki.freecad.org/Draft_AutoGroup).

What a twist!
Honestly, I'm shocked!!!
It seems that this wonderful opportunity is so unnoticeable that if it weren't for your message, I would never have known about it.
I've been struggling with all the objects appearing at the bottom of the tree view for over a year now!
There's clearly a design flaw somewhere.

kaiwas, 2025-02-08:
Yes^100!!!

kaiwas, 2025-02-08:
> with [Draft AutoGroup](https://wiki.freecad.org/Draft_AutoGroup).

everything turned out not as good as I thought.
this algorithm does not imply convenient construction of new objects inside the layer. at least not always
https://github.com/FreeCAD/FreeCAD/discussions/19480

yorikvanhavre, 2025-02-10:
Indeed all these features I think are there, but some are still buggy, and others are not clearly documented or easy to find by users. I propose we close this issue, and open new ones for each specific issue that is encountered.

Having smaller, well-defined issues is a big step forward because it allows other developers to spend less time understanding the problem and what needs to be done, and it allows for a quicker resolution.