FreeCAD/FreeCAD #14743: [UI] :  Some icons appear partially hidden at startup

luisDias89, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The icons for "New Part," "New Assembly," etc., appear partially hidden, and the general images of the projects below are not being displayed.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37739 (Git) Snap 1053
Build type: Release
Branch: main
Hash: ed672c5cd9560039756ddeaf4b12e4ad6ad6b8ca
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-06-17:
I've seen this happening a few times on my system, although I don't have steps to reproduce it. IIRC it tended to happen more often when switching themes.

Unrelated, but your screenshot seems to indicate you're also experiencing https://github.com/FreeCAD/FreeCAD/issues/13676

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37731 (Git) Snap 1052
Build type: Release
Branch: main
Hash: 6623b605282aa5ae61ae3df0adff2f6c857477fa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
```

maxwxyz, 2024-06-17:
@MisterMakerNL FYI

MisterMakerNL, 2024-06-17:
This bug is there since the beginning of the startpage!
It happens when you switch theme's.
And it stays like that for-ever.
@chennes

FEA-eng, 2024-06-29:
It seems to be the same problem as reported here: #14347

closed 2024-06-29 by maxwxyz