FreeCAD/FreeCAD #14745: All the issues in the new stylesheets/themes.

MisterMakerNL, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am going to use this spot to gather all the issues about the new stylesheets.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-17:
- https://github.com/FreeCAD/FreeCAD/issues/14747

MisterMakerNL, 2024-06-17:
This needs to be implemented 
https://github.com/FreeCAD/FreeCAD/pull/14592

kadet1090, 2024-06-17:
Tested with Qt6 and Fusion as Qt Style.

1. Disappearing down arrow icon in toolbars:
[image]

2. Unnecessary padding for toolbar icons:
[image]

3. Hard to read text in hovered menu entries due to low contrast:
[image]

4. No padding for tool buttons in the status area:
[image]

5. Unnecessary border for toolbars in QStatusBar and QMenuBar:
[image]
[image]

6. hovering over table rows highlights only cell:
[image]

7. Unreadable Combo Box list due to low contrast:
[image]

8. Controls could use a bit more horizontal padding to look better and be more readable (https://medium.com/@manonpiette.pro/margins-and-padding-e4e9d0efa65d):
[image]

9. Checkboxes are quite small - it would be wise to make them a bit bigger.
[image]

10. Radio buttons are not rendered well:
[image]

11. Spacing for buttons in spinboxes is uneven:
[image]

12. Hovering over treeview in overlay mode hovers whole row (probably Qt6 issue, see: https://github.com/FreeCAD/FreeCAD/issues/13760):
[image]

13. Buttons in task panel have broken laoyut:
[image]

14. Expression icon is not aligned vertically:
[image]

15. Scrollbar border is uneven (look at top button):
[image]

16. Yellow for marking "clicked" button in overlay panels seems very odd:
[image]

17. Icon for expanding tree branch is too big:
[image]

18. Images for theme selection buttons are not centered:
[image]

19. Strange black bar close to few drop-down toolbar buttons:
[image]

maxwxyz, 2024-06-18:
And there are still two dark themes. Behave has to be deleted too, you can only choose between two on the start page.

MisterMakerNL, 2024-06-18:
> And there are still two dark themes. Behave has to be deleted too, you can only choose between two on the start page.

https://github.com/FreeCAD/FreeCAD/pull/13772#issuecomment-2175012895

furgo16, 2024-06-18:
Thank you for the work in simplifying and upgrading the default themes. Some things I've noticed:

1. Toolbar separators are too dark on the Light theme. They might also need some horizontal padding: 
   [image]
1. Notification window is clipped horizontally:
 [image]
1. Active button highlight is nearly invisible. In this example, <kbd>Cancel</kbd> is active. The default Qt theme overlays a transparent color over the button's body to make it more clear:
   [image]
1. Unnecessarily colored checkbox when using keyboard navigation and highlighting the active checkbox. This also applies to checking a checkbox (the checkbox itself is blue when clicking on it):
   [image]
1.

MisterMakerNL, 2024-06-18:
Seems to be that the overlay isn't correct in classic/no-stylesheet.
https://forum.freecad.org/viewtopic.php?p=766453#p766453

MisterMakerNL, 2024-06-18:
Although it's gonna be annoying to do but I want to suggest renaming the theme's too:

- Freecad Dark
- Freecad Light
- Freecad Classic

I made this up myself and definitely didn't steal this idea from Blender.

furgo16, 2024-06-18:
I'd agree, for consistency. Although please consider also:

- Classic
- Dark
- Light

Why:

1. Simpler, shorter, straightforward to translate
2. It already conveys the essential information about the theme
3. Adding the "FreeCAD" suffix does not convey any additional necessary information. The user is using FreeCAD and changing the preferences to switch themes, so it's already clear to them that they are looking at a FreeCAD-specific theme.

This is of course just my opinion, feel free to disagree!

MisterMakerNL, 2024-06-18:
I can see why Blender did it. You say adding Freecad doesn't add any information, but it actually says it parts of Freecad. 
And with the option to mix in theme's from addons, it is useful to give this information.

kadet1090, 2024-06-18:
I'm in favor of adding `FreeCAD` prefix to "official" themes. Going with "Light" / "Dark" / "System" would be fine if we would not provide ability to create custom themes. With custom themes it will help distinguish the official themes from 3rd party ones.

MisterMakerNL, 2024-06-18:
> Seems to be that the overlay isn't correct in classic/no-stylesheet. https://forum.freecad.org/viewtopic.php?p=766453#p766453

Fixed it by setting an overlay stylesheet https://github.com/FreeCAD/FreeCAD/pull/14749/commits/84391e28c68194217ad7054f5264bd1ff2b5e951

maxwxyz, 2024-06-18:
I'm in favor still dropping the Behave theme. Otherwise anyone can do a PR and add his own theme to FreeCAD and it gets bloated fast. So 3 official ones, easier to maintain and anyone is free to create one by himself and distribute it through the addon manager. 
Otherwise I could change one color of dark slightly, make a PR with the theme and call it Darker and if more people do this we end up at the situation we have in the stable.

pierreporte, 2024-06-18:
Also in favor of dropping the theme. FreeCAD should concentrate on two carefully designed ones and the classic theme (at least in medium term for this one).

onekk, 2024-06-18:
Applying a stylesheet need sometimes a repaint action, I've discovered when fiddling with them, at least on Qt5, I have published on FreeCAD forum a post some time ago with python code that will trigger the repaint action.

Sadly not at my PC for some time to post here the relevant code (three lines in Python).

Maybe in a couple of days I will post here these lines.

MisterMakerNL, 2024-06-19:
Radio buttons reported by @kadet1090 should be fixed now, I boosted all icons sizes since I saw artifacts on lower resolution.
Only the Qsint icons stay the same. 
Should look like this now:
[image]

[image]

MisterMakerNL, 2024-06-21:
Fixed the overlay icons as mentioned by @kadet1090 
[image]
Although only for the light theme atm.

MisterMakerNL, 2024-06-21:
Fixed the spinbox buttons in light
[image]

MisterMakerNL, 2024-06-22:
Some small fixes:
[image]

furgo16, 2024-06-23:
@MisterMakerNL thanks. Do you think it would be possible to submit a draft PR, so that we can help with testing the changes?

onekk, 2024-06-23:
Hello, I've found the code now:

This function will do the magic when applying "role" to labels, i.e a dynamic property as described in:

Customizing Using Dynamic Properties[¶](https://doc.qt.io/qtforpython-5/overviews/stylesheet-examples.html#customizing-using-dynamic-properties)

`
def set_wd_prop(wid, prop, val):
    """Set Qt Widget property."""
    wid.setProperty(prop, val)
    wid.style().unpolish(wid)
    wid.style().polish(wid)
`

`
set_wd_prop(label, "role", "edit")
`

With this stylesheet (I have many of them for different 'roles')

`
            "QLabel[role='edit'] {"
            "    font-weight: bold;"
            "    font-style: normal;"
            "    text-decoration: none;"
            "    border: 2px outset gray;"
            "    border-radius: 5px;"
            "    padding: 5px;"
            "}"
`

The point is that the property change is not applied so you have to force it, probably the same is not working on some other things, maybe the force code above, (That I have probably converted from a C code, sadly, an accident in my computer, has deleted all my saved Forefox bookmarks, so I can't present the saved informations from where I have taken the code.)

Hope it maybe help in some cases.

Kind regards

Carlo D.

MisterMakerNL, 2024-06-23:
> @MisterMakerNL thanks. Do you think it would be possible to submit a draft PR, so that we can help with testing the changes?

https://github.com/FreeCAD/FreeCAD/pull/14749
The lag fix is btw from kadet https://github.com/FreeCAD/FreeCAD/pull/14820

MisterMakerNL, 2024-06-24:
Text select seems to be broken.

MisterMakerNL, 2024-06-24:
I renamed the theme's too Freecad Dark/Light/classic.
Also re-ordered startpage so the active theme is first doesn't fix this issue but improves it: https://github.com/FreeCAD/FreeCAD/issues/14819
[image]

MisterMakerNL, 2024-06-28:
Fixed missing text select color:
https://github.com/FreeCAD/FreeCAD/pull/14749/commits/5c422ad3b377c9b3012f3c2a96289a9a06778eaa

MisterMakerNL, 2024-07-01:
Borders are back in classic, not sure if this is causes by something I fixed:
[image]

maxwxyz, 2024-07-01:
can confirm

MisterMakerNL, 2024-07-01:
ok that was a stupid mistake.
I reorganized the code a bit but forgot a piece, in the light/dark overlay theme.

maxwxyz, 2024-07-01:
could you all check if your mentioned issues in the comments are fixed with the recent merges and add new issues for the ones which are still open? TIA!

pierreporte, 2024-07-01:
I’ll do it once the next weekly is available.

johnsonm, 2024-07-01:
The issue I had reported was fixed :+1:

furgo16, 2024-07-03:
> could you all check if your mentioned issues in the comments are fixed with the recent merges and add new issues for the ones which are still open? TIA!

I've updated the status of outstanding and fixed issues at: https://github.com/FreeCAD/FreeCAD/issues/14745#issuecomment-2175916382

@maxwxyz I'm happy to file separate issues as suggested, but I'm not sure what the best course of action is. Last time I attempted to file a bunch of issues at once, it was frowned upon. Perhaps the original description of this issue could be updated with a task list of pending bugs? Please advise. Thanks!

MisterMakerNL, 2024-07-03:
AS also mentioned above the custom colors are not set when booting freecad without settings, if you open the settings and click the colors then they work:
[image]

Syres916, 2024-07-15:
After the merge of https://github.com/FreeCAD/FreeCAD/commit/3a32ae9985475416324551f44a92f3e578964cf2 there's a couple of issues I've picked up on and got fixes for. Currently the TechDraw Proj Group task panel looks like:

[image]

and with https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:FC_Dark_Light_TD_Fixes applied it looks like:

[image]

willemx, 2024-07-16:
In the CAM module in Dark mode, when I double-click an operation to inspect its settings, I get this:

[image]

In Classic mode it looks like this:

[image]

MisterMakerNL, 2024-07-16:
Looking at the overlay icons the posts are from before my last pr.
I wanted to check it out myself but CAM isn't working on my build because libraries aren't complete.

willemx, 2024-07-16:
issue is also still in build 38231.

Syres916, 2024-07-16:
@MisterMakerNL they are smaller but still twice as big as I would have expected and on (personal view) the wrong side, something like:

[image]

```
QToolBox::tab {
  background-color: transparent;
  border: 0px solid #696969;
  color: white;
  background-image: url(qss:images_classic/Arrow-down-white-small.png);
  background-repeat: none;
  background-position: right;
}
```

MisterMakerNL, 2024-07-16:
This was actually a good bug, indeed Qtoolbox is broken as hell.
I managed to fix it tho. I ended up removing the arrow, there is also no arrow in classic so I see no need to try and make it work with the scaling issues.
Looks now like this, it's not perfect but I don't understand who some frames are missing a background.
[image]
biggest issue was:
QToolBox  QScrollArea  QWidget  QWidget {
Which overwrites all qwidgets in Qtoolbox and had to be:
QToolBox > QScrollArea > QWidget > QWidget {

MisterMakerNL, 2024-07-16:
Fixed it here:
https://github.com/MisterMakerNL/FreeCAD_MisterMaker/commit/634a05c7c9b0a4cb541955d5c2a3581fc70d1b9e
Also did some clean-up.

MisterMakerNL, 2024-07-16:
Made new PR here: https://github.com/FreeCAD/FreeCAD/pull/15471

Syres916, 2024-07-17:
> Made new PR here: #15471

There's an issue on Linux with `padding-left: -17px;` in `QToolBox::tab`, the tab titles get chopped for example **Layout** ends up looking:

[image]

I assume you've not forgotten about the typos for `check-checked-`_color_`.png` which should be `checkbox-checked-`_color_`.png`

MisterMakerNL, 2024-07-17:
> > Made new PR here: #15471
> 
> There's an issue on Linux with `padding-left: -17px;` in `QToolBox::tab`, the tab titles get chopped for example **Layout** ends up looking:
> 
> ![CAM_PR15471](https://private-user-images.githubusercontent.com/46537884/349511040-e4c278f3-fdad-41db-99db-e0473c397228.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyMzYzMzMsIm5iZiI6MTcyMTIzNjAzMywicGF0aCI6Ii80NjUzNzg4NC8zNDk1MTEwNDAtZTRjMjc4ZjMtZmRhZC00MWRiLTk5ZGItZTA0NzNjMzk3MjI4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE3VDE3MDcxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE3NjJkMjlkNGJjODdiNjUxZmRlOTFhM2JjNzUzMzk2MTM0ZDM3MWM0NGFhYjM3MWJjMzZlYjgxMGZkOTY2Y2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vC_33nuw1_WOgNszw41gUrN9GB5FceejaqhdCUfqTm4)
> 
> I assume you've not forgotten about the typos for `check-checked-`_color_`.png` which should be `checkbox-checked-`_color_`.png`
That padding is annoying! I added it because in windows I had a 17px empty space.
It looks like this for me:
[image]
The typo is defintly a stupid name. I got checkbox-checked and got a single check-mark. Which is called check-checked, but that doesn't make much sense.

Syres916, 2024-07-17:
> That padding is annoying! I added it because in windows I had a 17px empty space.

I extracted the latest AppImage and made the changes to just check whether it was my distro's Qt version but the same chopping of the start of the word happens using the AppImage as well.

MisterMakerNL, 2024-07-17:
> > That padding is annoying! I added it because in windows I had a 17px empty space.
> 
> I extracted the latest AppImage and made the changes to just check whether it was my distro's Qt version but the same chopping of the start of the word happens using the AppImage as well.

I'll remove it then, rather have a good readable version in linux then no empty space in windows.

MisterMakerNL, 2024-07-17:
The icon in the title bar is also 17px, so I guess that is the cause of this.

MisterMakerNL, 2024-07-19:
PR is ready here for anyone who want to check it out:
https://github.com/FreeCAD/FreeCAD/pull/15471
It's only stylesheet fixes so you can just download the stylesheets too.
They only work with the last weekly.

marcuspollio, 2024-07-22:
Hi @MisterMakerNL 

Following today's PR #15471 merge, here are some issues on the Start page :

- _Light theme_ has black selection color and black text in the Welcome/quick start dialog (maybe fixed by PR #15574 ?) :

[image]

- _Light theme_ and _Dark theme_ lost the hover properties on New file and Examples. Hover work on the _Classic theme_.

-  _Dark theme_  has too low contrast between background and text on on New file and Examples :

[image]

Syres916, 2024-07-23:
> Following today's PR #15471 merge, here are some issues on the Start page :
> 
>     * _Light theme_ has black selection color and black text in the Welcome/quick start dialog (maybe fixed by PR [[Start] Fix ThemeAccentColors not populated... #15574](https://github.com/FreeCAD/FreeCAD/pull/15574)) :

With PR15574 as soon as the user clicks on either theme in the ThemeSelector it will check for the existence of the accentcolor parameter and apply on unfound as per:

[image]

I realised that if the user had set them to their own personal colors but not clicked Done previously then they would be reset to default which would have caused some negative user feedback.

Syres916, 2024-07-23:
@MisterMakerNL FreeCAD Light appears to be missing a block for the QColorDialog, currently:

[image]

after applying:

```
QColorDialog QWidget {
  background-color: #f0f0f0;
  color: black;
  selection-background-color: @ThemeAccentColor1;
  selection-color: white;
}
```

[image]

MisterMakerNL, 2024-07-23:
@Syres916 The filenames are readable as the correct background is applied after restart.
The hover and pressed state are in the Stylesheet, as suggested. But don't seem to work.
Code is here:
https://github.com/FreeCAD/FreeCAD/blob/a71f49f4f61a3849e69a99daf19c79f19431889d/src/Mod/Start/Gui/FileCardDelegate.cpp#L46
Code in stylesheet is here:
https://github.com/FreeCAD/FreeCAD/blob/a71f49f4f61a3849e69a99daf19c79f19431889d/src/Gui/Stylesheets/FreeCAD%20Dark.qss#L2789

MisterMakerNL, 2024-07-23:
Fixed the Qcolor thingy here, which was weird it changed to arrow color to the correct color...
Thanks for you help!
https://github.com/FreeCAD/FreeCAD/pull/15586

Syres916, 2024-07-23:
> @Syres916 The filenames in start are readable as the correct background is applied after restart. The hover and pressed state are in the Stylesheet, as suggested. But don't seem to work. Code is here:
> 
> https://github.com/FreeCAD/FreeCAD/blob/a71f49f4f61a3849e69a99daf19c79f19431889d/src/Mod/Start/Gui/FileCardDelegate.cpp#L46
> 

I believe it's more 'dead' code because the Start module initialises seconds before the Gui is actually visible to the user. I'll get a solution but for 1.0 it might have to be done via the cfg file but we'll see.

MisterMakerNL, 2024-07-23:
@Syres916 IF we can fix the background not updating then at least it will be readable, the hover over isn't that necessary.

Syres916, 2024-07-24:
This is the best I can do before release, not sure whether I'm hitting a Qt6.7.x bizarre bug or what but the lookup for the state isn't working for the QListView, by all means give it a try:

https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Start_FileCard_Color_Fixes

Ona seperate note, there's something wrong in the FreeCAD Light stylesheet, I've eliminated the cfg file but only on Qt6.7.2 does the Start subwindow tab jump from the bottom left to bottom middle like it thinks it's on Windows 11!! FreeCAD Dark and Behave-dark are unaffected so I'll have a look tonight to see if I can find a missing or rogue entry.

Syres916, 2024-07-24:
I've sorted the FreeCAD Light Start tab issue, it was all down to amount of cutting and pasting must have got some text messed up so all good now with the changes in #15586

closed 2024-07-29 by chennes