FreeCAD/FreeCAD #14746: non standard 3dcopnnection implementation

moolight-seashell, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

the Y/X axis are swaped when using a 3d mouse

i found an old fix here : https://forum.freecad.org/viewtopic.php?t=72880

but the setting page no longer exist ?
because i use the open driver spacenavd on linux i cannot set a axis settings for each software so this is anoying

juts reverse the x y axis by default with the spacemouse device

### Full version info

```shell
0.21 & weekly branch
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

3dconnection inputs / hid

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

moolight-seashell, 2024-06-17:
edit ok i hav found the setiings here 
https://wiki.freecad.org/Interface_Customization

but make this settings checks by default =)

maxwxyz, 2024-06-17:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/11632

closed 2024-06-17 by maxwxyz