FreeCAD/FreeCAD #14747: Checkbox state not visible in new Dark/Light themes

dapetcu21, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the Dark/Light themes in the weekly build (on macOS ARM64), the checkbox visuals look the same for both the checked and unchecked states, so it's impossible to tell if a checkbox is checked or not.

Additionally, there are a few non-critical issues with the theme that make it awkward to visually parse (see the notes in the below screenshot).

[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-17:
@MisterMakerNL FYI

MisterMakerNL, 2024-06-17:
Yeah, it's not only the state it are all images! Maybe we can rebuild the weekly, so this isn't going to be posted a gazzilion times? :)

maxwxyz, 2024-07-01:
@dapetcu21 is this still an issue for you?

dapetcu21, 2024-07-01:
Hi! The icons and checkboxes are indeed fixed in the latest nightly. All the other minor issues I pointed out in the above screenshot are still present.

MisterMakerNL, 2024-07-01:
> Hi! The icons and checkboxes are indeed fixed in the latest nightly. All the other minor issues I pointed out in the above screenshot are still present.

What? really? I though most of them where caused by missing images. Could you post a new screenshot?

dapetcu21, 2024-07-01:
[image]

MisterMakerNL, 2024-07-01:
You DPI settings seem to cram everything together. How does it look with no-stylesheet?

dapetcu21, 2024-07-01:
These are the default DPI settings on a 13" macbook air. I assume that by no-stylesheet you mean the "Classic" theme:

(Also something I forgot to add, in the dark theme screenshot, setting rows that have a checkbox are not aligned with the labels of the other entries in the settings menu)

[image]

dapetcu21, 2024-07-02:
Would you prefer I make separate issues for each one of these?

maxwxyz, 2024-07-02:
If they still exist and have no issue, yes please.

MisterMakerNL, 2024-07-02:
It could use some improvements, but there are currently bigger fish to fry than optimization for a 13" screen!

dapetcu21, 2024-07-02:
Fair enough. I just plugged in a 4k screen. The scrollbar in settings issue did actually resolve itself. I'll close this and open separate issues for everything that's left.

closed 2024-07-02 by dapetcu21