FreeCAD/FreeCAD #14751: BIM: Error, the walls move with the carpentry

gianlucafiore, 2024-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A recurring bug that has happened to me for a long time since I have been working with freecad, is that when I position a window on a wall, if I want to move it to correct its height of the sill or distance to the edge of the wall, with the command move (M,V), the first time I move it I am always moved by the wall as well,  I run Ctrl+Z, it puts everything back in place, I move the window again, and it works correctly without moving the wall.

1. Walls with a simple window
[image] 

2. Moving the window only
[image] 

3. Host wall move
[image] 

I thought maybe it's a configuration issue, but I can't find anything to fix it

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-06-18:
I confirm.
It seems that this has not happened before.

The wall moves with the window once (the very first). Then, if you delete the window and create the same one, the problem no longer appears.

https://github.com/FreeCAD/FreeCAD/assets/32273833/db6b1928-7aa0-4bc8-a88e-709a3a08f09e

maxwxyz, 2024-06-24:
@yorikvanhavre FYI

yorikvanhavre, 2024-06-25:
I cannot reproduce this? I did the following:

* Click the wall tool, draw 2 points
* Select the wall
* Switch to front view
* Click the window tool
* Place the window

So far so good. Then

* Move the window

Result: the window moves, the wall does not. Anything you guys doing differently?

Or by script:

```python
trace=Part.LineSegment(FreeCAD.Vector (-100.0, 100.0, 0.0),FreeCAD.Vector (1700.0, 100.0, 0.0))
wall = Arch.makeWall(trace,width=100.0,height=2000.0,align="Left")
face = FreeCAD.ActiveDocument.Wall.Shape.Faces[2]
pl = DraftGeomUtils.placement_from_face(face, vec_z = wp.axis)
pl.Base = FreeCAD.Vector(300.0, 0.0, 500.0)
win = Arch.makeWindowPreset('Open 1-pane', width=800.0, height=800.0, h1=3.0, h2=3.0, h3=3.0, w1=3.0, w2=3.0, o1=3.0, o2=3.0, placement=pl)
win.Hosts = [FreeCAD.ActiveDocument.Wall]
# ...
Draft.move([FreeCAD.ActiveDocument.Window], FreeCAD.Vector(-300.0, 0.0, 0.0), copy=False)
FreeCAD.ActiveDocument.recompute()
```

gianlucafiore, 2024-06-27:
Can you try moving this wall in this test file? to the left or right

https://drive.google.com/file/d/1niGEQ2GpDB_2XavfyG6aC4Jb-HlG2zEH/view?usp=drive_link

[image]

yorikvanhavre, 2024-07-01:
There is something weird in that file, that I can't figure out...

Indeed if I move either of these windows, the wall moves together. However if I do Ctrl+Z afterwards, the windows don't come back to their previous location but somewhere else.

Even weirder: If I don't save the file, but use File->Revert, then after that the bug is gone

gianlucafiore, 2024-07-01:
Yes, it has happened to me with some files a few times in my life when I use freecad, and I can't be sure exactly when this problem starts to happen 😵‍💫

Roy-043, 2024-07-02:
When you open the file some objects need to be recomputed. And if you do a full recompute first the problem does not occur. This and Yorik's experience suggest that perhaps the delayed loading of certain modules plays a role here.

yorikvanhavre, 2024-07-11:
Delayed loading or not, it's weird that the file opens with "to be recomputed" objects... Normally nothing should be computed on file load

gianlucafiore, 2024-08-09:
[Sin nombre.zip](https://github.com/user-attachments/files/16567550/Sin.nombre.zip)

I'm using the latest weekly build, and I still have the same problem :(

gianlucafiore, 2024-08-09:
@yorikvanhavre  Apparently the problem originates when I modify software settings, when I reset it to default it stops happening

maxwxyz, 2024-08-10:
Could you narrow it down which settings are causing this?

luzpaz, 2024-10-23:
@gianlucafiore have you run into this again ?