FreeCAD/FreeCAD #14759: CAM: Request for change of FreeCAD's default Slitting Saw shape file because of its unnatural reference point

jkvfc, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD's default Slitting Saw shape file has a strange reference point:

[image]

The reference point should be on the underside of the saw blade (think about tool presetting) then you get a perfect simulation:
[image]
@sliptonic ,
Please could you replace file slittingsaw.fcstd with the attached file slittingsawnewreference.fcstd.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37735 (Git)
Build type: Unknown
Branch: main
Hash: 2a6a3649aff0eb5c6a54e67899ed5cc5c6332265
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * MyScripts
```

### Subproject(s) affected?

CAM/Path

### Anything else?

[slittingsawnewreference.zip](https://github.com/user-attachments/files/15883397/slittingsawnewreference.zip)
[TestToolsTappingSlittingCorrect.zip](https://github.com/user-attachments/files/15883398/TestToolsTappingSlittingCorrect.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-18:
Related: https://github.com/FreeCAD/FreeCAD/issues/9466

jkvfc, 2024-06-19:
@maxwxyz 
I think not related to #9466. Only a simple file exchange by @sliptonic  is needed.

sliptonic, 2024-06-19:
Yes.  Simple change.  I'll get to it but it's a fantastic issue for a first time contribution, even from a non programmer.  It's just editing a FreeCAD project file.  The bottom center of the sketch should be on the origin.

jkvfc, 2024-06-20:
@sliptonic,
Will open a PR. Need a bit of time to study the procedure. Please could you take a look at my second open issue #14765. Thanks a lot.

commit 75fc19ea (2024-06-24): CAM: Set new reference point for default slitting saw tool - fix #14759
closed 2024-06-24 by sliptonic