FreeCAD/FreeCAD #14760: Preferences need cosmetic fix

Kuzma30, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]

### Full version info

```shell
latest dev
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-06-18:
@Kuzma30 could you include your full version info? I believe you are using [OpenTheme Light](https://github.com/obelisk79/OpenTheme), which is a 3rd party theme. The built-in Classic theme looks like this:

[image]

If that is the case, you might need to file the bug against the OpenTheme project.

I have also noticed the horizontal scrollbar on that screen. It seems it's there as the default width of the window is not wide enough to contain all widgets.

furgo16, 2024-06-18:
The invalid height mentioned on this issue is, I believe https://github.com/obelisk79/OpenTheme/issues/60. It should have been fixed by now, but I've not checked, as I no longer use OpenTheme.

Kuzma30, 2024-06-18:
Same on Classic
[image]

```
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37817 (Git)
Build type: Debug
Branch: master
Hash: 4a11652263c8635da0be55e41c0a5a9d04767dcf
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.2.dev
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * sheetmetal 0.4.18
  * ose-piping
  * dodo 1.0.1
  * fasteners 0.5.22
  * FeedsAndSpeeds 0.5.0
  * ose-3d-printer-workbench 0.1.0
  * OpenTheme 2024.5.16
  * Assembly3 0.12.2
  * addFC 0.3.1
  * ose-workbench-core

```

furgo16, 2024-06-18:
Of all the issues reported, I can only reproduce the one about the horizontal scrollbar:

[image]

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37821 (Git)
Build type: Unknown
Branch: main
Hash: 1ccb74f141ec4d1276b39d18fb99f276d09e007e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
```
Have you tried with clean configuration files? I've found out that once you've had the OpenTheme applied, switching back to Classic does not reset all changes: https://forum.freecad.org/viewtopic.php?t=87293

Kuzma30, 2024-06-18:
Just remove all config files and get same error with height.
I have Display with 1920*1080 resolutions.

maxwxyz, 2024-06-18:
@kadet1090 FYI

kadet1090, 2024-06-26:
@maxwxyz can you assign me? Also this should be considered a bug

maxwxyz, 2024-07-30:
@kadet, the scrollbar on the preferences is still there, so seems not to be fixed.
@Kuzma30 could you confirm?

kadet1090, 2024-07-30:
I am no longer able to reproduce - so I'll need more info.

maxwxyz, 2024-11-08:
closing for no feedback, I cannot reproduce either.

PR #15652 (2024-07-29): Gui: Cosmetic Fixes
This PR fixes various small cosmetic problems with preferences, like wrong padding, inconsistent alignment, horizontal scrollbar that should not be present etc.

Examples of before and after:
| Before | After |
| -- | -- |
|[image] | [image] |
| [image] | [image] |
| [image] |[image] |
| [image] | [image] |

Fixes: #14972 
Fixes: #14760 
Fixes: #15308
closed 2024-11-08 by maxwxyz