FreeCAD/FreeCAD #14762: FreeCAD current and OpenCascade 7.8.1

JKBert, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

I use FreeCAD as a rolling release on a Linux Devuan wokstation (testing). This morning, I have upgraded my source tree to upgrade FreeCAD and build aborts with:

[ 48%] Building CXX object src/Mod/Fem/Gui/CMakeFiles/FemGui.dir/TaskCreateElementSet.cpp.o
/home/bertrand/git/FreeCAD/src/Mod/Fem/Gui/TaskCreateElementSet.cpp:40:10: fatal error: Standard_math.hxx: Aucun fichier ou dossier de ce type
   40 | #include <Standard_math.hxx>
      |          ^~~~~~~~~~~~~~~~~~~

libocct-foundation-dev (7.8.1) doesn't contains Standard_math.hxx.

Last week, with the same OpenCascade release, I have successfully upgraded FreeCAD.

Best regards,

JB

### Full version info

```shell
Build stage aborts, here is output of last git:

8f6798fcc1..2ca9b6ef80  main       -> origin/main
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-06-18:
Yes, I would add, I had to manually modify this file also to compile _FreeCAD_ with _OCCT 7.8.1_:

https://github.com/FreeCAD/FreeCAD/blob/2ca9b6ef8036d5fd808f01205b66eb0424059e21/cMake/FindOCC.cmake#L162-L164

Though, I don't even remember if I succeeded in this.

Syres916, 2024-06-18:
I just removed the `#include <Standard_math.hxx>` from _src/Mod/Fem/Gui/TaskCreateElementSet.cpp_ without changing any other file and it compiled fine.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37820 (Git)
Build type: Release
Branch: main
Hash: a36a60d29dfe376c42576bbf4ed7c3fc7e9850a8
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * SketchArch
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1

```
I'll ask the author of https://github.com/FreeCAD/FreeCAD/pull/14756 to remove the line as part of that PR.

Syres916, 2024-06-19:
OK, I'll create a PR in the next hour or so.

Syres916, 2024-06-24:
@JKBert this can be closed as per https://github.com/FreeCAD/FreeCAD/pull/14766

maxwxyz, 2024-06-24:
I'll close it then