FreeCAD/FreeCAD #14764: Undoing helix causes segfault

mpfusion, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating a subtractive helix followed by “undo” causes a segfault. Steps to reproduce:

- open attached file 
[helix-segfault.tar.gz](https://github.com/user-attachments/files/15886417/helix-segfault.tar.gz)

- select VThreadProfile
- Part Design WB → Subtractive Helix

  Helix parameters:
  Axis: base Z azis
  Pitch: 1.5 mm (FC freezes, but hang on, it just takes forever)
  Height: 10 mm
  → OK

- click undo → segfault

```
SubtractiveHelix: gp_Dir() - input vector has zero norm
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f3fc80bd050]
#1  0x7f3f066a03b4 in PartDesignGui::ViewProviderBody::unifyVisualProperty(App::Property const*) from /home/mp/usr/bin/freecad-bin/lib/PartDesignGui.so+0x184
#2  0x7f3f066a5ba8 in PartDesignGui::ViewProviderBody::onChanged(App::Property const*) from /home/mp/usr/bin/freecad-bin/lib/PartDesignGui.so+0x28
#3  0x7f3fca5ec985 in App::Property::hasSetValue() from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0x125
#4  0x7f3fca66b7aa in App::PropertyMaterialList::setValues(std::vector<App::Material, std::allocator<App::Material> > const&) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0xca
#5  0x7f3f066a52a7 in PartDesignGui::ViewProviderBody::copyColorsfromTip(App::DocumentObject*) from /home/mp/usr/bin/freecad-bin/lib/PartDesignGui.so+0xe7
#6  0x7f3f066a594c in PartDesignGui::ViewProviderBody::updateData(App::Property const*) from /home/mp/usr/bin/freecad-bin/lib/PartDesignGui.so+0x1bc
#7  0x7f3fcb34d5ff in Gui::ViewProvider::update(App::Property const*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x5f
#8  0x7f3fcb368a55 in Gui::ViewProviderDocumentObject::update(App::Property const*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x55
#9  0x7f3fcae86cfa in Gui::Document::slotChangedObject(App::DocumentObject const&, App::Property const&) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x3a
#10  0x7f3fca491504 in boost::signals2::detail::signal_impl<void (App::DocumentObject const&, App::Property const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::DocumentObject const&, App::Property const&)>, boost::function<void (boost::signals2::connection const&, App::DocumentObject const&, App::Property const&)>, boost::signals2::mutex>::operator()(App::DocumentObject const&, App::Property const&) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0x1f4
#11  0x7f3fca4bbc32 in App::DocumentObject::onChanged(App::Property const*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0xf2
#12  0x7f3fca5ec985 in App::Property::hasSetValue() from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0x125
#13  0x7f3fca589df1 in App::TransactionObject::applyChn(App::Document&, App::TransactionalObject*, bool) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0x91
#14  0x7f3fca58a043 in App::Transaction::apply(App::Document&, bool) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0xf3
#15  0x7f3fca45452a in App::Document::undo(int) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0xca
#16  0x7f3fcae8473a in Gui::Document::undo(int) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x4a
#17  0x7f3fcb300a69 in Gui::View3DInventor::onMsg(char const*, char const**) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x4b9
#18  0x7f3fcae1e789 in Gui::Application::sendMsgToActiveView(char const*, char const**) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x39
#19  0x7f3fcaf3f251 in Gui::Command::_invoke(int, bool) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x281
#20  0x7f3fcaf3f6f3 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x133
#21  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f4f) [0x7f3fc8784f4f]
#22  0x7f3fc921b782 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#23  0x7f3fc921e3ab in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xab
#24  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x256e8a) [0x7f3fc9315e8a]
#25  0x7f3fc9315fd7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#26  0x7f3fc940eaaa in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa
#27  0x7f3fc9263db8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a8
#28  0x7f3fc9221fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#29  0x7f3fc922a552 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x13c2
#30  0x7f3fcaee3c18 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x68
#31  0x7f3fc874d6f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#32  0x7f3fc922865e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1be
#33  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1bdbd8) [0x7f3fc927cbd8]
#34  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c0f60) [0x7f3fc927ff60]
#35  0x7f3fc9221fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#36  0x7f3fcaee3c18 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x68
#37  0x7f3fc874d6f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#38  0x7f3fc8b2c3ed in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x80d
#39  0x7f3fc8b00cac in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#40  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x6deca) [0x7f3fc230ceca]
#41  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x299) [0x7f3fc64ea7a9]
#42  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x54a38) [0x7f3fc64eaa38]
#43  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f3fc64eaacc]
#44  0x7f3fc87a5836 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#45  0x7f3fc874c17b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#46  0x7f3fc87542d6 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#47  0x7f3fcae2b846 in Gui::Application::runApplication() from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x756
#48  freecad-dev(+0x8f0e) [0x5576b3ffcf0e]
#49  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7f3fc80a824a]
#50  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f3fc80a8305]
#51  freecad-dev(+0x9211) [0x5576b3ffd211]

```

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37748 (Git)
Build type: Release
Branch: main
Hash: 8f6798fcc1914889bfdad1b83f5a66d8651733d2
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Installed mods: 
  * ThreadProfile 1.90.0
  * BOLTSFC
  * kicadStepUpMod 11.2.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-27:
@FlachyJoe any chance you can look at this ?

luzpaz, 2024-08-25:
@FlachyJoe soft bump

FlachyJoe, 2024-08-25:
@luzpaz I don't see anything related to helix in the backtrace but I'll take a look when I'm done with Pad UpTo*.

luzpaz, 2024-09-24:
Should this be a blocker ?

FlachyJoe, 2024-09-24:
I can't reproduce with recent dev. @mpfusion could you please test with v1.0RC1?

mpfusion, 2024-09-24:
I just tried to reproduce with a current b38c42262f58cf0587b627e7a866d3756da34dc4 but it doesn't segfault any more. I get different errors, though. But that's not what this report is about. I can confirm the issue is fixed on Version: 1.1.0dev.38830 (Git).

luzpaz, 2024-09-24:
@mpfusion do you need to open a ticket for the different errors ?

Closing as fixed.

closed 2024-09-24 by luzpaz

mpfusion, 2024-09-25:
Done: #16812