FreeCAD/FreeCAD #14769: Sketcher - Polar transformation explodes sketch.

GAZ082, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As per the video, trying a Polar transformation wreaks havoc.

Could not replicate with simpler sketches. Example attached.

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves.backup1713713759.0032098 0.6.29 (Disabled)
  * QuickMeasure 2022.10.28
  * ExtremeProDark 2.9.0
  * Curves.backup1693059780.2922277 0.6.13 (Disabled)
  * CfdOF.backup1710694906.6805208 1.24.9 (Disabled)
  * Assembly4.backup1696800837.713216 0.50.2 (Disabled)
  * Curves.backup1714093860.7275996 0.6.34 (Disabled)
  * OpenTheme 2024.5.16
  * CfdOF.backup1699129791.898633 1.24.7 (Disabled)
  * Assembly4.backup1710902627.637761 0.50.9 (Disabled)
  * Defeaturing.backup1710696629.0288966 1.2.2 (Disabled)
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/100194/2d8b08da-9ce7-4b61-92da-f8a40df04b38

[Example.zip](https://github.com/user-attachments/files/15888754/Example.zip) - Change Zip to FCStd

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-18:
@PaddleStroke @pieterhijma FYI

PaddleStroke, 2024-06-19:
Thanks for the report. I'll check it out.

PaddleStroke, 2024-07-03:
@GAZ082 can you please tell me how you made the arcs of this sketch?
Because the arcs are Clockwise which generate the issue you have.
I have done many test and cannot find how you made such an arc. Using Arc, fillet, polyline all create CCW arcs automatically.

PaddleStroke, 2024-07-03:
The bug actually happens because when the rotate tool creates the rotate arc, it creates is CW as it currently is. But once you validate the tool, the geometry is added to the sketch and sketcher automatically switch it to CCW. Which generates the failure you have.

GAZ082, 2024-07-03:
The sketch was not directly generated. It was taken from a face of an imported STEP file using Shapebinder.

GAZ082, 2024-07-03:
Just brainstorming here, if the direction of the arc impacts the algorithm and you know how to find out the direction of the arc, cant you just adjust the part of the algo to adapt to the direction of the arc?

PaddleStroke, 2024-07-03:
> The sketch was not directly generated. It was taken from a face of an imported STEP file using Shapebinder. The process was: STEP > Shapebinder > Convert to Sketch.

I think there might be some issue in the 'convert to sketch' command. There seems to be something wrong with the arc but I can't pinpoint what is wrong.

@wwmayer do you perhaps know what is going on here? See the simplified file in the comment above.

PaddleStroke, 2024-07-04:
I have done some more tests: 
- Translate and scale tools both result in the same behavior as rotate : the arc jumps around.
- The old 'rectangular array' tool does not jump arround.

Apparantly the way the geometries are added with both tools are different: 
- The old tool is adding the new geometries directly into the sketch object
- The new tools are using the python converter that creates the geometries through python commands. And the python commands that creates arcs are fixing automatically the bad arcs. Hence the jump.

GAZ082, 2024-07-20:
Hey  @PaddleStroke , any news? Thanks.

nazar-pc, 2024-08-18:
I have seen this flipping/buckling frequently with rectangles, arcs (very often in cases with slot shape) that is much easier to reproduce, but I suspect has the same nature:

https://github.com/user-attachments/assets/98e9d210-301f-4510-ad5e-8b17128249a2

It happens in cases where there are technically more than one way to satisfy the solver based on specified constraints alone and it seems to pick whatever it likes the most, which sometimes is logical (but wrong) result and sometimes makes no sense at all (on the first glance at least). Above demo demonstrates both.

I'm not exactly sure how to solve this algorithmically, but it is a pain for fully dynamic models where the size can vary in a wide range depending on a spreadsheet property.

If performance was of no concern than it seems that updating value in tiny increments (1 mm in my demo above) and doing recompute after each update would in fact work around the issue, but with some of my sketches already taking many seconds to recompute on changes that is not applicable as the default approach.

luzpaz, 2024-10-11:
bumping for visibility

closed 2024-10-28 by yorikvanhavre

GAZ082, 2024-10-28:
Awesome, yesterday while working on a project had this issue again :) Thanks @PaddleStroke

maxwxyz, 2024-10-31:
@GAZ082 could you test if the issue is resolved in the last weekly?