FreeCAD/FreeCAD #14770: Can Fasteners Properties be updated to Title case?

leoheck, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Properties of Fasteners Workbench are lower case. This could be improved to mach Freecad style. 

However it looks like the Workbench cannot be improved alone since it may break old files as the author said. Maybe Freecad could improve the way those properties are parsed ignoring the case or when displaying convert them to the Title case.

See the discussion here:
https://github.com/shaise/FreeCAD_FastenersWB/issues/391

What do you think?

### Full version info

```shell
I am considering the 0.22-dev but it is not tied to a single version.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-06-20:
Fasteners Workbench was updated already, it looks like it is working. There is also the SheetMetal workbench that could receive the same treatment, and maybe others too.

shaise, 2024-06-20:
Right now the way freecad display the properties is by taking the programmatic name and split it based on hyphens, underscore and case-ness. so 'my-parameter' becomes `my parameter` and 'myParameter' becomes `my Parameter`.
So, a simple solution, if not letting the programmer use a custom name, is to automatically Title Case all versions for the sake of uniformity.    
In this case both 'my-parameter' and 'myParameter' will be shown as `My Parameter`

leoheck, 2024-06-20:
What if Freecad could add an optional parameter to have a Name for the property to be displayed as it is, when this is not set, then it uses the programmatic name as it is doing right now but converting what has to be converted and also using Title Case.

shaise, 2024-06-20:
This is best, but I don't think it will be done very soon as there are much more urgent things. Converting to Title Case however, is rather an easy fix.

leoheck, 2024-06-20:
Totally. Freecad is just getting polished and attracting a lot of attention especially on this pre-1.0 release. These small things make the tool look better and more trustfull.

hasecilu, 2024-07-01:
As shaise already did, the "FreeCAD" way to name the **Property** variables is by using PascalCase notation, the same applies for **Parameters** (those to save WB preferences).
One tiny "**pythonic**" nitpick is that following that the variables linting color in the code will be the one of a class instead of getting the color of a variable. As you can see a very minor rant. This could be fixed allowing to external Python WB to use camelCase (or even snake_case) for property variables and still getting that special treatment for the label.
[image]

> What if Freecad could add an optional parameter to have a Name for the property to be displayed as it is, when this is not set, then it uses the programmatic name as it is doing right now but converting what has to be converted and also using Title Case?

If done should also consider translations. On Qt there is support to translate property titles, property names and property tooltips. In the case of properties current translation support fails when property have 2 or more "words" (example: "Magnet Option" should be "Opción Imán") FreeGrid WB.

[image]

luzpaz, 2024-07-12:
Maybe this feature can be renamed to more appropriately name the issue which is independent of Fasteners workbench but can apply to any 3rd party workbench.