FreeCAD/FreeCAD #14772: Random color problem

Pendelf, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using Random Color functions, the color changes only if you save the file and open it afterwards

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.21
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.5.1
  * Pyramids-and-Polyhedrons
  * sheetmetal.backup1704622370.7006474 0.3.12
```

### Subproject(s) affected?

None

### Anything else?

https://youtu.be/vvO53QYUJ2Q

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pendelf, 2024-06-18:
Also, further editing resets the color to the initial color (added a link to the video)
https://youtu.be/UPrq8Mh-mvk

maxwxyz, 2024-06-18:
@davesrocketshop FYI

davesrocketshop, 2024-06-18:
Verified. The color is set, but not displayed. I'm on it...

commit aac8bdd7 (2024-07-04): Fixes #14772: Random color problem
closed 2024-07-04 by wwmayer