FreeCAD/FreeCAD #14775: Align to selection problem

Pendelf, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the Alight to selection function, the camera rotates at a strange angle

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.21
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.5.1
  * Pyramids-and-Polyhedrons
  * sheetmetal.backup1704622370.7006474 0.3.12
```

### Subproject(s) affected?

None

### Anything else?

https://youtu.be/980MIwEzisA

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-18:
What is the intended behavior? How should FreeCAD know the rotation the user currently likes?
@Rexbas FYI

Rexbas, 2024-06-18:
Right now, the camera always rotates to the same orientation for a given face. However, depending on the current camera orientation, the camera could move a lot to align to the face. This is not always the most optimal camera movement. I am planning to work on improving this in the next couple of weeks.

PR #20088 (2025-03-22): Gui: Align to selection using a smaller rotation
Fixes #14775.

Previously, the align to selection tool always rotated to one specific orientation for a given face/edge. This could result in a very wild disorienting rotation. Now it looks at the current camera orientation and chooses between one of four orientations to reduce the amount of rotation. It is now also more likely that a straight edge is aligned horizontally or vertically but it is not guaranteed. In another PR I want to do horizontal or vertical alignment based on the longest straight edge in a face. But that is still work in prog [... cut]
closed 2025-03-22 by chennes