FreeCAD/FreeCAD #14777: Addon Manager: No link to source / author

maxwxyz, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Split from: https://github.com/FreeCAD/FreeCAD/issues/14726
There is no link to the repository or whatever the source happens to be for the workbench for example to check the author, source code, etc. before installing it

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-20:
@chennes can you kindly add @hasecilu to the appropriate status so I can assign this ticket ?

chennes, 2024-09-20:
Invitation sent.

PR #16667 (2024-11-18): Addon Manager: Show repository URL if it exists on the metadata
@maxwxyz, is this what you were thinking?

~~Draft because links are not opened.~~

As you can see translation is being reused.

[image]

Fix: #14777
commit 4c1f94f6 (2024-11-18): Addon Manager: Show repository URL if it exists on the metadata

Fix: #14777
closed 2024-11-18 by chennes