FreeCAD/FreeCAD #14779: [Problem] Part Design "Shape appearance" (color) doesn't change

MiqCG, 2024-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you try to change the color of a Part Design body, with Properties panel > Shape Appearance, it doesn't change at all.

Steps to reproduce:
1. open the last revision of FreeCAD
2. Part Design > new body > new Sketch
3. Circle or rectangle
4. Pad the sketch
5. Select Body
6. Properties panel > Shape Appeareance
7. Select any color and accept
8. ... can you see any change?

Color can be changed in a Part WB part.

I guess is a regression from this PR #14615 for this issue #14568 because Part Design color works in revision 37730.

Don't know if it is related to this issue but if you save the file from revision 37819 and open it with 37730, the Report view says:
23:11:52  No extension found of type 'Gui::ViewProviderFaceTexture'
23:11:52  No extension found of type 'Gui::ViewProviderFaceTexture'
23:11:52  No extension found of type 'Gui::ViewProviderFaceTexture'

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.22
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-19:
@davesrocketshop FYI

MiqCG, 2024-07-06:
Looks like no Pull Request is related to this issue but in Appimage 38043 is again possible to change the body color to a specific one (not only with Random color) and Transparency is working too.

davesrocketshop, 2024-07-06:
I'm also not able to reproduce from the latest source build. I was able to reproduce it when it was first submitted, but I don't know what changed.

MiqCG, 2024-07-06:
Mystery
Anyway, is one blocker bug less to 1.0 :-)
Would you like this issue to be closed?

davesrocketshop, 2024-07-06:
I guess so. It can always be refiled later

closed 2024-07-06 by MiqCG

madpipeline, 2024-08-05:
In dev build: 
```
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git) AppImage
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Romanian/Romania (ro_RO)
```

I'm experiencing this behavior:

* After assigning a random color, opening the `Appearance...` menu resets the color.
* Setting the color from the `Custom appearance` -> `Ambient color` works. The part's faces are now colored.