FreeCAD/FreeCAD #14784: BIM: Draft_Offset and BIM_Offset2D are confusing

marcuspollio, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the BIM Workbench, one can perform offsets of selected edges and faces with 2 tools: Draft_Offset and BIM_Offset2D (based on Part 2D Offset).
They provide different and complementary functionalities, yet they both produce offsets of supported geometry.
Moreover, their use and interaction differ quite significantly, making the experience confusing :

1) Both tools only work for "Draft" geometry (wires, faces, etc.) but they are not greyed-out when non-supported geometry is selected (such as 3D/BIM objects or any other Part-based geometry), leading to error messages in the Report/Notification. The same applies to the Draft WB.

2) The Draft_Offset provides two modes: a simple offset on either side of the selected geometry, and a "OpenCascade-Style" offset. On a face/closed geometry, the OpenCascade-Style mode only allows to offset outwards, whereas the simple mode allows both inwards and outwards offsets.
Draft_Offset makes use of the 3D view with interactive feedback following the user cursor, allowing snapping.

3) The BIM_Offset2D also provides two modes and three join types: a "Skin" mode similar to the simple offset of Draft_Offset and a "Pipe" mode somewhat similar to the OpenCascade one. However, depending on the join type chosen, different geometries are produced. All BIM_Offset2D allow an inward offset of a face/closed geometry.
Because it's based on the Part 2D Offset tool, BIM_Offset2D does not make use of the 3D view and feedback of the created offset geometry is only available with the "Update View" option.

Ideally, only one Offset tool should be proposed to the user, with its underlying modes and options contextually defined depending on the selection. However, this necessitates significant changes that go beyond the BIM WB alone.
A first improvement would be to solve the first aspect above. For the remaining two, this requires further discussion.

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS (DE: Gnome/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-06-20:
Of course I agree with you, and I also still use one or another depending on the case.
I would suggest modifying the Draft tool so it can produce a Part 2D offset. Basically:

- The tool still works like the current Draft tool
- The task panel gets additional options to replicate the Part 2D offset settings
- If those are used, a Part 2D offset object is created

How does that sound?

marcuspollio, 2024-06-25:
Yes, it makes sense.

I don't know how complex this endeavor could turn out, so I would propose for the time being, as a quick fix, to add conditions to Offset tools to become inaccessible (greyed out) when not supported geometry is selected :
Draft_Offset and BIM_Offset2D/Part_Offset2D work with wires and faces, but are greyed out with Shapes.
Part_Offset3D/BrepOffset works with Shapes/BIM 3D elements, but is greyed out with wires and faces.