FreeCAD/FreeCAD #14786: Cannot reset a Revolute joint to its original position once moved

furgo16, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open the attached file with FreeCAD: [F-02-C_bug.zip](https://github.com/user-attachments/files/15893942/F-02-C_bug.zip) (it's a truncated version of a window assembly. For simplification purposes, please ignore the fact that there is no real hinge)
1. Ensure you have the <kbd>CAD</kbd> navigation style activated. Not strictly necessary, but it's one where you don't need modifier keys to move along joints.
1. Double-click the Assembly object in the Tree view to activate the assembly
1. In the 3D view, click somewhere in an area where there is no object. Not strictly necessary, but it unselects the whole assembly, which simplifies the view.
1. In the 3D view, place the cursor over the `Outer stile left003` component
1. Left-click and keep the button pressed
1. Move the cursor left or right to move (manually animate) around the Revolute joint in the assembly
1. Stop pressing the left button. The `Outer stile left003` component is rotated around the Revolute join at an angle that is not the original
1. **Expected**: there is a way to accurately reset the Revolute join to its original position
2. **Actual**: the above is not possible:
   - Re-doing steps 3 to 8 allows you to move the component along the joint again, but there is no numerical indication of the rotation as you are manually doing the move (the Property view is blank). You can only guess and do the repositioning by trial and error (moving, selecting component, reading Property view until you get to a position that is accurate enough).
   - Selecting the `Outer stile left003` component in the Tree view and then manually editing the <kbd>Placement</kbd> > <kbd>Angle</kbd> property fails in one of two ways: 1) inputting the new value, the component rotating as expected, then hitting Ctrl-R, and then the value automatically going back to the last value 2) inputting the new value and then the value automatically going back to the last value

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37731 (Git) Snap 1052
Build type: Release
Branch: main
Hash: 6623b605282aa5ae61ae3df0adff2f6c857477fa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.5
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-25:
Indeed there currently no way to put a revolute joint to it's original position. Same for other joints.
By original position do you mean 0 deg between the JCS, or do you want to be able to set a specific value?

furgo16, 2024-06-25:
@PaddleStroke thanks. Indeed, if the original position was at 0 degrees, I meant being able to return to 0 degrees. Although being able to set a specific value would also take care of the issue (I would set the angle to 0°).

Another option might be to snap to the original position when rotating around the vicinity of it.

furgo16, 2024-06-26:
As an alternative, looking at https://github.com/FreeCAD/FreeCAD/pull/14738, setting a lower limit to 0° would also work.