FreeCAD/FreeCAD #14787: Attachment offset in sketch using function returns 0

chopsywa, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I use a function to create an attachment offset in a sketch, the function has the correct value in the dialog, but results in 0 as the value of the offset. Screen grabs below. 

[image]

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Curves.backup1706353084.2536438 0.6.13
  * fasteners.backup1718262130.4318058 0.5.22
  * OpenDark.backup1718262281.5371034 2024.3.13
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-19:
Is it just a bug when using sketches?

chopsywa, 2024-06-19:
It seems to be. I am using formulas in other data areas OK.

Syres916, 2024-06-19:
I cannot replicate the bug and I've attached a file showing inside and outside of a Body it's working as expected using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37820 (Git)
Build type: Release
Branch: main
Hash: a36a60d29dfe376c42576bbf4ed7c3fc7e9850a8
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * SketchArch
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1

```
[Issue14787.zip](https://github.com/user-attachments/files/15901731/Issue14787.zip)
Replace the zip extension with FCStd

chopsywa, 2024-06-21:
I loaded your file and it worked OK. I found the issue. It is the program itself and not related to the files.
If I have an operation, such as a pad, or boolean, or anything except a sketch **open**, any changes I make to the properties of a sketch are reverted when the operation is closed, or cancelled. If I use a function for one of these properties, the property has a value of zero as per my screen grab above and when I cancel the operation, it reverts to the prior value.

Now here is the issue. If I have multiple files open and I leave the operation window open in one project, this issue happens across **all** projects. I can double click on a pad in one drawing, then move across to your Issue14787 document, any data properties I make in your document are reverted when I exit the operation back on the original document. Also,  your document will have a 0 value for the function in the same situation.

The bottom line is if you are working on multiple document, you **must** ensure you don't leave any document mid operation as it will affect other documents.

Does this make sense?

Syres916, 2024-06-21:
I understand, never done that in the seven years of using FreeCAD but it's a legitimate report. The first thing I need to do is try to build a Unit Test in Python to demonstrate the situation so the senior developers have something to run their code against. To give you context, the project is in a feature freeze and this could be classed as a big change so best to assume it won't be In version 1.0 and if it is class as a bonus.

chopsywa, 2024-06-21:
Sounds good. It doesn't concern me. It was by chance that I had done it and I am just reporting everything I come across as I assume this is what the developers want coming up to release.

FEA-eng, 2025-09-17:
[This](https://github.com/FreeCAD/FreeCAD/issues/14787#issuecomment-2182007908) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-02-17:
[This](https://github.com/FreeCAD/FreeCAD/issues/14787#issuecomment-2182007908) is still relevant

Perhaps #21978 could fix it

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260210 (Git shallow)
Build date: 2026/02/10 18:10:36
Build type: Release
Branch: (HEAD detached at ecbe21ca0)
Hash: ecbe21ca03a59723c90ce5a677ae7ccea06ebc12
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-08-27:
> If I have an operation, such as a pad, or boolean, or anything except a sketch **open**, any changes I make to the properties of a sketch are reverted when the operation is closed, or cancelled.

This is still relevant.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260825 (Git shallow)
Build date: 2026/08/25 17:50:34
Build type: Release
Branch: (HEAD detached at 964fce2ed)
Hash: 964fce2edc9a7e1905c2fd3fcd6e7be90afc8375
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```