FreeCAD/FreeCAD #14788: The Behave-dark theme has a particularly wide sidebar

chwt163, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the Behave-dark theme, the sidebar of the TechDraw workbench is particularly wide

[image]

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * Behave-dark-1 0.0.5
  * A2plus 0.4.65
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-19:
I believe this has already been fixed since the release you are using, see screenshots:

[image]

[image]

you can try by replacing your existing Behave-dark.qss with this one:
[Behave-dark.zip](https://github.com/user-attachments/files/15901484/Behave-dark.zip)
replace zip with qss as the extension.

chwt163, 2024-06-19:
> I believe this has already been fixed since the release you are using, see screenshots:
> 
> 
> you can try by replacing your existing Behave-dark.qss with this one: [Behave-dark.zip](https://github.com/user-attachments/files/15901484/Behave-dark.zip) replace zip with qss as the extension.

Sorry, I didn't describe it in enough detail. It works fine on Windows, the problem happens on Linux.

Syres916, 2024-06-19:
And as you see from the screenshots they are from Linux Mint (the _lm_ logo in the bottom left corner), full info:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33839 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: 1380323d8562578487d03091d3e2f572ac1d20ef
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * SketchArch
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1
```

Are you saying you've tested the file I provided and it makes no difference on your distribution?

chwt163, 2024-06-19:
> And as you see from the screenshots they are from Linux Mint (the _lm_ logo in the bottom left corner), full info:
>
> ```
> OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
> Word size of FreeCAD: 64-bit [...]

Yes, on my manjaro system, I tested it and it is still wide, and manually turning it down will not show it completely

Syres916, 2024-06-20:
Ok, so it's likely to be a Qt6.7.x issue and I'll need to find a way to build a reliable VM based on Qt6.7.1, there's other issues reported on the forum with 6.7.0 that I'm not going to waste time on that version.

Syres916, 2024-06-20:
Something to try while I work on setting up a new VM is to edit the stylesheet file and at the end add:

```
QWidget[objectName="TechDrawGui__TaskProjGroup"] QPushButton {
    min-width: 24px;
}
```
You will probably have to adjust the 24px to make it exactly as you want but it should have an effect either way.

chwt163, 2024-06-21:
> Something to try while I work on setting up a new VM is to edit the stylesheet file and at the end add:
> 
> ```
> QWidget[objectName="TechDrawGui__TaskProjGroup"] QPushButton, QWidget[objectName="TechDrawGui__TaskProjGroup"] QWidget#wbList QPushButton, QWidget#wbList QWidget[objectName="TechDrawGui__TaskProjGroup"] QPushButton, QWidget[objectName="TechDrawGui__TaskProjGroup"] QSint--ActionGroup QFrame[class="content"] QToolButton, QSint--ActionGroup QFrame[class="content"] QWidget[objectName="TechDrawGui__TaskProjGroup"] QToolButton, QWidget[objectName="TechDrawGui__TaskProjGroup"] QWidget#wbList QSint--ActionGroup QFrame[class="content"] QToolButton, QWidget#wbList QSint--ActionGroup QFrame[class="content"] QWidget[objectName="TechDrawGui__TaskProjGroup"] QToolButton, QWidget[objectName="TechDrawGui__TaskProjGroup"] QSint--ActionGroup QFrame[class="content"] QWidget#wbList QToolButton, QSint--ActionGroup QFrame[class="content"] QWidget#wbList QWidget[objectName="TechDrawGui__TaskProjGroup"] QToolButton {
>     min-width: 24px;
> }
> ```
> 
> You will probably have to adjust the 24px to make it exactly as you want but it should have an effect either way.
> 
> Edit: More enhanced modification for the stylesheet file

This modification has little effect.

I found the reason for the problem, because the 'padding' value of 'QPushButton' is too large, change line 1483 of the theme qss file, change it to:
`padding: 4px 6px;`

closed 2024-06-21 by chwt163

Syres916, 2024-06-21:
@chwt163 thanks for the constructive feedback. I won't be able to submit a pull request for such a global change as it could affect for example a German language user having text compressed or chopped when two buttons are side-by-side. I can go through and check as many task panels as possible and apply the two settings to them and ensure there are no regressions.
On another subject, please see https://github.com/FreeCAD/FreeCAD/pull/14827 I'm sure you have thoughts on the proposed removal of the stylesheet. Feel free to post in the PR comments, mine will be posted later this weekend when I've got more time.