FreeCAD/FreeCAD #14795: testPartDesignElementMapSketch: Duplicate test method name

chennes, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The linter is complaining that testPartDesignElementMapSketch already exists.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37710 (Git)
Build type: Release
Branch: main
Hash: a6937d8d2677594f548af218cc92f797d9a475d9
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-06-20:
The duplicated function name actually hides a failing test case.

chennes, 2024-06-21:
@bgbsww is this one of yours?

PR #14801 (2024-07-01): PD: Rename duplicated test method
Fixes #14795: testPartDesignElementMapSketch: Duplicate test method name
commit 3abd1989 (2024-07-01): PD: Rename duplicated test method

Fixes #14795: testPartDesignElementMapSketch: Duplicate test method name
closed 2024-07-01 by chennes