FreeCAD/FreeCAD #14796: Dev Version: CAM wb | Sanity check issues

ShamanTcler, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Two issues:

1) While performing a sanity check the model gui area was updated 3-4 times moving the view space around
2) When it came time to save the file it went to my $HOME dir, not the dir where the actually FreeCad partfile was. Normally I would store the report with the part files. 

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
  * Curves 0.6.39
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-18:
@ShamanTcler can you compress and attach an example file please ?

ShamanTcler, 2024-07-18:
I was talking to another developer; it seems issue 1 is because the sanity checker has embedded graphics in it and Sliptonic could only get the images by man handling the screen ... works for me.

Issue 2 is simply an issue of resetting the file manager dialog to offer to store the "sanity" files in the directory of the CAD file, not the directory where FreeCad was installed.

PR #15934 (2024-09-10): CAM Fix minor sanity bugs
set default directory for sanity
Respond correctly if user cancels  
Fixes #14796
commit 6eefc8e6 (2024-09-10): set default directory for sanity (#15934)

Respond correctly if user cancels  Fixes #14796
closed 2024-09-10 by sliptonic