FreeCAD/FreeCAD #14798: [Problem] Assembly WB: linked bodies don't respect imposed joint limits

MiqCG, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Looks like Joints Limits of Integrated assembly WB doesn't work properly with Linked bodies.
When assembled bodies dragged, plenty of this message appear in the Report View "Assembly : Ignoring bad solve, a grounded object moved."
See attached folder with a Test file and the linked files.

I have checked a similar example with "Insert components" and works properly. 

Don't know if this pull request #14738 will affect or solve this problem. I have searched into Github [is:issue label:"WB Assembly" limits ] with no more relevant results

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.22
```

### Subproject(s) affected?

Assembly

### Anything else?

[Github_Ass_Range.zip](https://github.com/user-attachments/files/15903755/Github_Ass_Range.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-21:
Please provide a video or steps how to reproduce.

MiqCG, 2024-06-21:
Wow, I thoroughly check the problem before reporting and now I cannot reproduce it. 
Now the limit to the joints are respected.
Two days ago it was pretty obvious to me so I didn't record it and deleted the test files. 
Strange but better. You can close this issue.

Sorry for the inconvenience.

closed 2024-06-22 by MiqCG