FreeCAD/FreeCAD #14799: Undoing a PartDesign migration crashes

Jookia, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening a document with an empty body and part design box then trying to do a mirror prompts to migrate from a legacy PartDesign. After migrating and creating the transform I have the prompt for a mirrored box in a body.

If I abort the prompt and try mirroring again, I get a crash.
If I continue the prompt, undo, then create a body using the part design box I get a crash.

Here's a file that contains the un-migrated objects:  [PartDesignTransformed2.zip](https://github.com/user-attachments/files/15903446/PartDesignTransformed2.zip)

Here's a gdb debug session: [debug.txt](https://github.com/user-attachments/files/15903781/debug.txt)

I did some code reading and it looks like despite undoing a transaction the active part is still remembered. This can be semi-confirmed by 'Gui.activeView().getActiveObject('pdbody')' causing the crash after aborting a mirror.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git)
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.67
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-06-19:
I can not reproduce
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37827 (Git)
Build type: Debug
Branch: main
Hash: d1d2b730ef62642bee333ea36fdfb92885fedb72
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
```

Jookia, 2024-06-19:
So you're able to abort the mirror operation then try again successfully?

FlachyJoe, 2024-06-19:
No crash but an error `Mirrored: No originals linked to the transformed feature.`

Jookia, 2024-06-19:
Interesting. Could you run "Gui.activeView().getActiveObject('pdbody')" after aborting the mirror operation? What does it say?

Jookia, 2024-06-20:
I built with address sanitizer and got this log confirming it's a use after free: 
[asan.txt](https://github.com/user-attachments/files/15909485/asan.txt)

FlachyJoe, 2024-06-20:
Ok I got it. I don't know what I did different before.

FlachyJoe, 2024-06-20:
I think the problem is that the Mirror cancellation cancels the migration too.
The doc workflow is kept `Workflow::Modern` due to storage in `WorkflowManager::dwMap` and so no migration request is done anymore.

Jookia, 2024-06-20:
How does that explain the use after free?

FlachyJoe, 2024-06-20:
Also `ActiveObjectList::_ObjectMap` is not reset when migration is undone.

Jookia, 2024-06-20:
Should it be explicitly reset? Looking at the code it seems like undoing a regular body creation doesn't reset it either.

Jookia, 2024-06-20:
So far I'm under the impression that ActiveObjectList::objectDeleted should be catching the pdbody and removing it, but that doesn't seem to be the case.

maxwxyz, 2024-08-11:
I'll close this as the migration tool was removed. Please let me know otherwise.

closed 2024-08-11 by maxwxyz