FreeCAD/FreeCAD #14802: Errors/Warnings/Msgs without any reference to track for the problematic object.

leoheck, 2024-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Random messages when loading (some old) files.

I would like to fix these issues, but these messages are generic and do not mention the object. I could list the objects like these, but this msg could have the Name/Label/Reference of the problematic object, right? Every message should refer the problematic item. Sometimes I navigate through the Report View and I see issues that I cannot fix or track since they are generic.

[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * sheetmetal 0.4.18
  * InventorLoader 1.4.0
  * fasteners.backup1718838499.3717518 0.5.23 (Disabled)
  * ose-piping
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * fasteners.backup1718509812.8005707 0.5.21 (Disabled)
  * OpenDark 2024.3.13
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.6.5
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-22:
@bgbsww is this related to TNP changes?

bgbsww, 2024-06-22:
I don't think so, but until the requested changes in visibility are made I don't think there is a way to tell.

closed 2025-07-20 by github-actions[bot]