FreeCAD/FreeCAD #14803: [ASSEMBLY] : FreeCAD closes unexpectedly when moving an assembler.

luisDias89, 2024-06-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was trying out the new weekly release and started to open the backhoe bucket assembly that comes standard in FreeCAD. When I try to move the bucket, at a specific point, FreeCAD terminates the process and exits.
Take a look at the video:

https://github.com/FreeCAD/FreeCAD/assets/48370099/27a57ca9-81b9-4b6a-8bbc-7153614bc333

https://github.com/FreeCAD/FreeCAD/assets/48370099/c15b4974-77cf-433b-8cf1-7a0c3b90952c

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luisDias89, 2024-06-20:
@PaddleStroke

PaddleStroke, 2024-06-21:
Aik siong is looking into it.

PaddleStroke, 2024-07-03:
@aiksiongkoh after the latest solver update I can still generate a crash while moving the backhoe around. Could you please take a look? Thanks !

PaddleStroke, 2024-07-18:
I think this is now fixed. I cannot get it to crash now.

luisDias89, 2024-07-19:
@PaddleStroke 

[image]
I tested it now, and it no longer crashes, but the solver is still very heavy, and the objects start getting all mixed up.

but this ticket seems to be resolved.
@maxwxyz

closed 2024-07-19 by maxwxyz