FreeCAD/FreeCAD #14807: PartDesign: Pad QListWidget signal not found

maxwxyz, 2024-06-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to pad a simple profile of 2 circles creates the warning:
`QObject::connect: signal not found in QListWidget
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-20:
@kadet1090 could this be related to https://github.com/FreeCAD/FreeCAD/pull/14433 ? I've just used the default values : Length and 10mm

Syres916, 2024-06-20:
Fix already merged https://github.com/FreeCAD/FreeCAD/commit/8f6798fcc1914889bfdad1b83f5a66d8651733d2

maxwxyz, 2024-06-20:
The weekly is from after this merge and still has this issue?

Syres916, 2024-06-20:
Nope, if you look down the list of assets in the weekly builds you'll notice the build revision hasn't increased for the Windows build because there was a compiling issue that Werner had to fix.

closed 2024-06-21 by maxwxyz