FreeCAD/FreeCAD #14818: empty recent files

kinghat, 2024-06-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when starting with a fresh freecad install you see an empty recent files block. i think it should be shown only in cases where there are recent files:

[image]

it looks out of place for a fresh start of the application.

### Full version info

```shell
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-06-20:
I think that the empty section is still useful, to indicate new users that it exist. But instead of emptiness, it would be better to add a short text explaining that no file were opened yet.

kinghat, 2024-06-20:
> I think that the empty section is still useful, to indicate new users that it exist. But instead of emptiness, it would be better to add a short text explaining that no file were opened yet.

i think this would be better than the current state, its just taking up space. if that is the desired situation then it should be below examples until there is an actual file.

if this issue is addressed: https://github.com/FreeCAD/FreeCAD/issues/13846 then their first and current document would be added very shortly and the recent files would appear and the user would see it.

luzpaz, 2024-08-02:
soft ping to @FreeCAD/design-working-group