FreeCAD/FreeCAD #14819: highlight initial theme state on fresh freecad install

kinghat, 2024-06-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when you start a fresh install of freecad you are presented with which theme you want via selection. the current situation doesnt make it obvious what the current state the app is in. on linux i believe its in "classic" theme but as you can see in the image that is not clear as my system is also using a dark mode:

[image]

it would be helpful to show a selection border around the theme of the current state of the app. 

### Full version info

```shell
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-06-20:
This gonna break when a custom theme is set.
Although not sure how you end up with a cust theme on first start.

kinghat, 2024-06-20:
> This gonna break when a custom theme is set. Although not sure how you end up with a cust theme on first start.

ya this is just for initial theme selection so it wouldnt apply.

luzpaz, 2024-07-18:
@MisterMakerNL is there a solution for this that you know of ?

luzpaz, 2024-11-07:
We should probably address this UI/UX issue before 1.0 release.

MisterMakerNL, 2024-11-08:
We could fix this if we name the thumbnails so they can be pointed at directly in the style-sheets.
The I could set "themepreview #dark" in the dark stylesheet to be highlighted. Do the same with light.
We can also add it too classic.
I would also add the others to every theme to reset them.
Hoover is also still broken btw.