FreeCAD/FreeCAD #14822: Tree view highlight state obscures content in Classic theme

johnsonm, 2024-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I am not hovering over the tree view, it is visible:

[image]

However, when I am hovering, the white-on-white is unreadable:

[image]

I do not think I have changed the settings for theme or colors from the default when I started:

[image]

This affects only the classic theme. All the other themes appear to work as intended.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37839 (Git)
Build type: Unknown
Branch: main
Hash: 9b547bb90fbdf5f9e6df17fee640566cbe2ab37e
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-06-22:
This only happens in activated overlay mode. If the task/tree view is a child of a dock window the issue doesn't occur.

maxwxyz, 2024-06-22:
@MisterMakerNL FYI

MisterMakerNL, 2024-06-22:
already fixed in my pr https://github.com/FreeCAD/FreeCAD/commit/84391e28c68194217ad7054f5264bd1ff2b5e951

please report more issues here: https://github.com/FreeCAD/FreeCAD/issues/14745

johnsonm, 2024-06-22:
Awesome, now that I know about #14745 I'll close this one as duplicate. Thanks!

closed 2024-06-22 by johnsonm