FreeCAD/FreeCAD #14825: Improving the usability of working with the workplane (suggestion)

kaiwas, 2024-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, this handy tool only works with an on-screen button, which can be very difficult to find among the other on-screen buttons.

The work plane selection key combination only works in the draft workbench. In order for this operation to work, I had to write macros and assign hotkeys to them.

My proposal is to do this at a “basic” level (macros are not obvious to the average user).

The video demonstrates the work.

https://github.com/FreeCAD/FreeCAD/assets/32273833/c0bf647e-c8cf-4d75-ae72-ab4677291e7d

The SHIFT+X / SHIFT+Y / SHIFT+Z keys move the work plane to zero coordinates, but set their direction according to the main planes.
The Shift+W key activates the work plane selection tool.
The Shift+Q key moves the selected object to the active work plane.

This approach is somewhat similar to working with a 3D cursor in Blender. This is one of the key and convenient features because it allows you to quickly move objects to the desired location (resembles, but is not, anchors)

Macros can be downloaded from this link. 
https://disk.yandex.ru/d/BDxPna5la6EZ-A

Perhaps they should be improved  (I don’t understand much about this). . But it would be better if all this was available at the “kernel” level.

Many thanks to everyone who helped me write the “move object to plane” macro. @Roy-043 , (qewer33 in discord)

### Full version info

```shell
any version
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-06-25:
That's interesting indeed.

In the BIM WB there are already 3 new shortcuts to set the WP to top, front and side positions (in the snapping menu). One could indeed add more. The object -> WP is interesting, I had never thought of that.. That would work pretty much like Blender

furgo16, 2025-05-13:
To summarize:

1. [DONE] Select work plane command: shortcut keys <kbd>W</kbd>, <kbd>P</kbd>
    > The Shift+W key activates the work plane selection tool.
    > The work plane selection key combination only works in the draft workbench. In order for this operation to work, I had to write macros and assign hotkeys to them.

   Now fixed with https://github.com/FreeCAD/FreeCAD/pull/21157 :-)
2. [DONE] Switch to default work plane commands: shortcut keys: <kbd>W</kbd>, <kbd>P</kbd>, <kbd>1</kbd>; <kbd>W</kbd>, <kbd>P</kbd>, <kbd>2</kbd>; and <kbd>W</kbd>, <kbd>P</kbd>, <kbd>3</kbd>
   > The SHIFT+X / SHIFT+Y / SHIFT+Z keys move the work plane to zero coordinates, but set their direction according to the main planes.

   Alredy implemented (see commands and shortcuts in the Snapping menu)
3. [TODO] Move to working plane command: shortcut key proposal: <kbd>W</kbd>, <kbd>M</kbd>
   > The Shift+Q key moves the selected object to the active work plane.

Is this a correct summary? And could you elaborate a bit more on how 3 should work? I'll look at the macro as well, but I think it'd be good to discuss the implementation in writing too.

furgo16, 2025-05-13:
I'd be interested in the use case for "Move to working plane". I myself would use it to quickly move a wall to another level, by activating the new level's working plane, and then executing the "Move to working plane".

But generally, I'd be interested to only change the wall's Z placement. The "Move to working plane" would move the wall to the right Z placement, but it would rewrite X and Y to the center of the active working plane, which would be undesired in this use case.

kaiwas, 2025-05-13:
> And could you elaborate a bit more on how 3 should work? I'm including your macro's code as well, but I think it'd be good to discuss the implementation in writing too.

If you are familiar with Blender, it is something like:
1) move the 3D cursor to the selected (in this case, the plane).
2) Move the object to the 3D cursor (to the plane)
The difference is that in Blender, the 3D cursor only takes into account the location. And the working plane also rotates the object.
Perhaps some option might be needed to be able to do both.
The 3D cursor can also be moved to the vertices (the working plane cannot)

[image]

We will not be able to take into account all the alignment problems (sometimes the working plane can have a strange direction), but in any case, we quickly move the object in this way. This is the main thing.

kaiwas, 2025-05-13:
> But generally, I'd be interested to only change the wall's Z placement. The "Move to working plane" would move the wall to the right Z placement, but it would rewrite X and Y to the center of the active working plane, which would be undesired in this use case.

I tried to implement this using macros, so that moving or copying would take into account the height of the floor.
And I managed to make objects move up.
But I couldn't make everything work correctly when a wall from the second floor would be copied to the first floor.
The problem is that objects that have a "base" object always have zero coordinates (the actual position is specified at the sketch level). This is just awful!

I'm waiting for the refactoring of walls and windows so that I can experiment with this.

kaiwas, 2025-05-13:
It would be great to have some kind of "move to level" command.

[image]

furgo16, 2025-05-13:
Thanks for the clarification. I think it might make sense to implement these separately, then;

1. Move to working plane command: shortcut key proposal: <kbd>W</kbd>, <kbd>M</kbd>. Moves the selected object to the active work plane.
2. Move to level: shortcut key proposal: <kbd>M</kbd>, <kbd>L</kbd>. Select a level and an object, execute the command to move object to the Z coordinate of the selected level, taking into account the object's previous Z coordinate, as per https://github.com/FreeCAD/FreeCAD/issues/14825#issuecomment-2876188139

Does this make sense?

> The problem is that objects that have a "base" object always have zero coordinates (the actual position is specified at the sketch/base coordinates). This is just awful!

For the "Move to level": Could this not be solved by checking whether the object has a base AND zero coordinates, and then modify the base's placement? If that conditon does not apply, then modify the object instead.

kaiwas, 2025-05-13:
> Does this make sense?

Oh, Yes! )

> For the "Move to level" command: Could this not be solved by checking whether the object has a base AND zero coordinates, and then modify the base's placement? If that conditon does not apply, then modify the object instead.

Perhaps this is a deeper task, but personally I really don't like that the object parameters are located at the base. The user always expects that the parameters of the object change at the object itself, and not at something located at a lower level in the view tree.

Honestly, I would prefer to never see all these "sketches" and other basic objects. But at the moment it is impossible. Although I came up with a way to hide them using macros)

furgo16, 2025-05-13:
Ok, then these two commands are something I can start looking into.

> Perhaps this is a deeper task, but personally I really don't like that the object parameters are located at the base. The user always expects that the parameters of the object change at the object itself, and not at something located at a lower level in the view tree.

I agree, I'm very aware of this behaviour, as it took me quite a while to figure it out. However, that's a bigger task that I feel should be tackled separately.

> Honestly, I would prefer to never see all these "sketches" and other basic objects. But at the moment it is impossible. Although I came up with a way to hide them using macros)

I guess this might be a preference that varies from user to user. I personally do not mind seeing them in the tree, as I have them collapsed most of the time.