FreeCAD/FreeCAD #14826: Strange behavior of the Trimex tool

kaiwas, 2024-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Trimex tool extrudes a plane normal to the surface. But to work correctly, you always need to monitor the direction of the working plane. Why?
In most cases, we need extrusion along the normal to the surface and unnecessary actions only complicate the process.

Maybe it would be correct to disable the work plane dependency and make such a dependency as an option.
This will make working with this tool easier.

Demonstration of strange behavior on video. Note that the correct result (which seems obvious) requires switching planes. For comparison, the behavior in Blender.

https://github.com/FreeCAD/FreeCAD/assets/32273833/4013d927-6995-426c-bda2-795da847dfbc

### Full version info

```shell
any
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-21:
The tool does use the mouse movement and can also use snaps. For this tool it is an issue that 'random' points in Draft are always projected on the current working plane.

kaiwas, 2024-06-21:
> The tool does use the mouse movement and can also use snaps. For this tool it is an issue that 'random' points in Draft are always projected on the current working plane.

Yes. I know this.

I assumed that it was possible to modify the extrusion tool (BIM_extrude) so that it would understand mouse movement and snaps, but now you can only enter values ​​into the fields.

In this case, Trimex could be left untouched and left for working with flat elements (lines, etc.)

yorikvanhavre, 2024-06-24:
I'm also wondering if this "Extrude" tool shouldn't be separated from Trimex... It does not make much sense to have it there

PR #18320 (2024-12-06): Draft: allow point for Trimex extrusion to be co-planar with the face
Fixes #14826.

If the point for the extrusion height is co-planar with the face, the normal of the face is used for the direction of the extrusion. The user can then enter a positive or negative value in the task panel. It is probably not the ideal solution, but better than what we have now I think.
commit e28286a2 (2024-12-06): Draft: allow point for Trimex extrusion to be co-planar with the face

Fixes #14826.

If the point for the extrusion height is co-planar with the face, the normal of the face is used for the direction of the extrusion. The user can then enter a positive or negative value in the task panel. It is probably not the ideal solution, but better than what we have now I think.
closed 2024-12-06 by yorikvanhavre