FreeCAD/FreeCAD #14830: Feature Request: call expression editor with on-view parameters

chrisb-github, 2024-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On behalf of forum user Dabbes:

### Full version info

```shell
In version 1.0 (0.22 dev), the On-View-Parameters are new in sketcher.
A value can be entered directly in the field that appears when drawing an object. Unfortunately, the equals sign ("="; key pressed) does not work in this field to call up the expression editor.
If, for example, dimensions and their alias designations are collected in a spreadsheet for the complete FreeCAD construction, these should be assigned by alias (e.g. "<<dimensions>>.length"; with "dimensions" name of the spreadsheet, "length" alias for the value length).
Unfortunately this does not work in the on-view parameters for now.

Please activate the expressions editor by "="-key in these on-view-parameter fields.
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://forum.freecad.org/viewtopic.php?p=767047#p767054

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

BanceFC, 2025-10-07:
No action on this? Is it not possible/difficult to do? Or is there no interest?
Not trying to pressure anyone just interested.

navycuda, 2026-01-12:
Please have a look at this. I tend to use very parametric heavy drawings and would like access to the expression editor during the creation of new geometry.