FreeCAD/FreeCAD #14838: Alias_Manager Macro Fails with errors in console

foxkit, 2024-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In version 0.21.2, but not in version 0.20.2, the Alias_Manager macro establishes no new aliases, and drops these  messages om the 
"Report view" window:
<code>
13:38:18  QImage: XPM pixels missing on image line 0 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 1 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 2 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 3 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 4 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 11 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 12 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 13 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 14 (possibly a C++ trigraph).
13:38:18  QImage: XPM pixels missing on image line 15 (possibly a C++ trigraph).
13:38:26  
Unable to complete task
</code>

I have completely uninstalled FreeCAD 0.20.2 and 0.21.2, and reinstalled them one at a time, reinstalling workbenches and macros.  The problem consistently occurs in 0.21.2 but not 0.20.2.  Both version dumps are included in the "Full Version Info" section below.

I can supply the spreadsheet file where I found the problem, but I tried making a new spreadsheet in v 0.21.2, and the problem still occurred.

### Full version info

```shell
In this version it fails:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * Alternate_OpenSCAD 1.0.0
  * dxf-library
  * FEMbyGEN 2.1.0
  * kicadStepUpMod 11.2.4
  * Lithophane
  * MakerWorkbench 1.0.1
  * parts_library
  * symbols_library
  * woodworking 0.21.2.33771
[/code]

In this version it works:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * Alternate_OpenSCAD 1.0.0
  * dxf-library
  * FEMbyGEN 2.1.0
  * kicadStepUpMod 11.2.4
  * Lithophane
  * MakerWorkbench 1.0.1
  * parts_library
  * symbols_library
  * woodworking 0.21.2.33771
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-21:
The macro you quote hasn't been maintained for 8 years, even the author doesn't use it as they use Realthunder's branch so there's no need for it. If you want a more up-to-date macro along the same lines, try https://wiki.freecad.org/Macro_EasyAlias
It's not for core developers to keep correcting the core code for old macros, it's the exact opposite.

foxkit, 2024-06-22:
Please excuse me.  I have not yet learned the boundaries of
responsibilities, nor did I understand that the problem reporting system
was only for problems known to be intrinsic in the core code.  I
incorrectly assumed that any problems reported would be dispatched to the
appropriate individuals.

Clearly, in this case I have been using a method that, although prevalent
on YouTube tutorials, is no longer supported.

I hope that my report was clear enough that you did not need to waste too
much time.

I will seek the alternative you suggested, and to the extent possible, will
add comments to places where I see that macro referenced.

I have been a software developer since ~1968.  I appreciate the difficulty
of evolving a code base.  I am impressed with the work you are doing.

Please pardon my breach of protocol.

Carl Mikkelsen
email: ***@***.***
email: ***@***.***
LinkedIn: //www.linkedin.com/in/carlmikkelsen/
<https://www.linkedin.com/in/carlmikkelsen/>

closed 2024-06-22 by Roy-043