FreeCAD/FreeCAD #14839: Property editor not is not detachable

Vincenzoferrara, 2024-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

the properties panel is integrated in models, but I would like them to be 2 separate views (that's how it was before).
up until now I had the separate panel, but after opening stable freecad, changing a setting and saving, it merged.
so I searched and I discovered that it is integrated in models, but if it is still detached for me, is there a setting?

ps:I would like there to be a search bar in the settings

link for doc: https://wiki.freecad.org/Property_editor

note: if it should be possible to detach it, it's a bug.

https://github.com/FreeCAD/FreeCAD/assets/64591523/e41469d3-f6d8-4330-a16e-1fcb4a289a16

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * AutoOpenSpreadsheet
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.5.16
  * Assembly4.backup1718498214.0742533 0.50.12 (Disabled)
  * fasteners 0.5.24
  * TabBar
  * Assembly4 0.50.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-06-21:
Try these settings
[image]
[image]

Vincenzoferrara, 2024-06-21:
perfect it works!!

closed 2024-06-21 by Vincenzoferrara