FreeCAD/FreeCAD #14841: Addon Manager is small in Qt6

Jookia, 2024-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On Qt6 the Addon Manager doesn't resize to fit the window, creating a very small viewport.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37842 (Git)
Build type: Release
Branch: jookia_testing
Hash: 0f9dcefb3f8a99969117ff60167e6a8fdb26706d
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.67

(Note: based on 9b547bb90fbdf5f9e6df17fee640566cbe2ab37e)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-06-22:
Well, I think, a screenshot would be good. Is it what you saw?

<details><summary>Screenshot</summary>

[image]

<details><summary>cmake configuration</summary>

```
cmake ../freecad-source -G Ninja                 \
-D CMAKE_BUILD_TYPE=Debug                        \
-D CMAKE_INSTALL_PREFIX=/home/xtemp09/sk-test/   \
-D CMAKE_CXX_STANDARD=20                         \
-D BUILD_ENABLE_CXX_STD=C++20                    \
\
-D CMAKE_C_FLAGS_DEBUG="-g -Og -fno-omit-frame-pointer -Wno-deprecated"   \
-D CMAKE_CXX_FLAGS_DEBUG="-g -Og -fno-omit-frame-pointer -Wno-deprecated" \
\
-D LINK_FLAGS_DEBUG="-fuse-ld=mold" \
\
-D Qt6_DIR=/usr/local/Qt-6.7.1/lib/cmake/Qt6 \
-D FREECAD_QT_VERSION=6
```

</details>

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37843 (Git)
Build type: Debug
Branch: main
Hash: 932c73f0de2d1e9ba3f60ae14d199cfe77255c3a
Python 3.10.12, Qt 6.7.1, Coin 4.0.2, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

Layout issues are the first things to be expected during the migration.

Jookia, 2024-06-22:
Yes, that's what I saw. However I don't have this with FreeCAD 0.21 with Qt6 so I imagine this is a recent issue.

xtemp09, 2024-06-22:
> FreeCAD 0.21 with Qt6

What version of _FreeCAD_ are you talking about? What version of _Qt 6_? What version of _Python_ do you have isntalled?

Jookia, 2024-06-22:
```
[code]
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.67
[/code]
```

PR #14978 (2024-06-27): Addon Manager: Allow widget expansion in Qt6
In Qt6 the Addon Manager does not expand to fill the view -- this PR addresses that. Fixes #14841.