FreeCAD/FreeCAD #14845: Measurement: Distance Free mode requires selecting points belonging to different faces/edges

FEA-eng, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[Measure distance](https://wiki.freecad.org/Std_MeasureDistance) was integrated into the unified measurement tool. However, a new limitation is that the (freely) selected points must belong to separate faces/edges. This is mentioned here: https://github.com/FreeCAD/FreeCAD/pull/13810#issuecomment-2120534645

Since it's caused by the fact that global selection is used by the tool, I guess that the only way to remove this limitation would be to disable the use of said global selection when the Distance Free mode is active (so it would work like old Measure distance).

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-23:
@hlorus FYI

masterxq, 2024-09-28:
I can confirm this. Additionally, it could be nice if the selected points would be highlighted again. Like the old tool did.

If there would be something that worked in the new measurement tool it would be really nice. But still missing the old simple one, I really loved that tool, for a lot of my workflows it was just perfect to get a very fast answer on a simple question. Maybe it can be reactivated? At least until this one is usable?

fredjohnston, 2024-10-20:
> I can confirm this. Additionally, it could be nice if the selected points would be highlighted again. Like the old tool did.
> 
> If there would be something that worked in the new measurement tool it would be really nice. But still missing the old simple one, I really loved that tool, for a lot of my workflows it was just perfect to get a very fast answer on a simple question. Maybe it can be reactivated? At least until this one is usable?
> 
> ```
> OS: Debian GNU/Linux trixie/sid (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 1.0.0RC2.38806 (Git) AppImage
> Build type: Release
> Branch: (HEAD detached at 1.0rc2)
> Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
> Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
> Locale: German/Germany (de_DE)
> Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
> Installed mods: 
>   * fcgear 1.0.0
>   * kicadStepUpMod 10.18.6
>   * Curves 0.6.8
>   * Dracula 0.0.5
> ```

TOTALLY AGREE!! PLEASE reactivate!

maxwxyz, 2024-10-21:
@hlorus can you guess how hard this is to implement?

FEA-eng, 2024-10-23:
@hlorus Can’t it work in the same way as the old tool (so pretty much just activating the old tool’s code when the Distance Free mode is selected)?

wwmayer, 2024-10-23:
> Somehow handle this in the selection by allowing multiple picking points per element. @wwmayer will have to say if that's OK to add.

This would add a lot more complexity to the selection mechanism compared to the complexity added to the measure tool.

> Storing a list of picked locations in the measure tool. This is not as elegant as currently the tool simply relies on the selection and having to store a separate list and keeping it in sync will add complexity.

The old distance measure tool didn't use the selection at all but added a callback function to the viewer and received the raw picked points from the 3d view. I would just do the same for the Distance Free measurement.

To avoid to add too much complexity to the TaskMeasure I would move the whole selection/picking logic to separate classes, i.e. an abstract base class and two sub-classes where one uses the selection and the other the picking mechanism.

luzpaz, 2025-06-18:
Added 'Solution proposed' label per wwmayer