FreeCAD/FreeCAD #14846: Help is not available in the Preferences

FEA-eng, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Help button was removed from the Preferences because it was non-functional, also in the 0.21.2 release. In 0.22, the Help module is merged. The next improvement could be to implement support for help in the Preferences. It should lead to proper sections of wiki pages describing selected preferences. 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-06-24:
You saw that with the new Help module there is now a "Help" tab in the General preferences, right? What would you say is missing there?

FEA-eng, 2024-06-24:
> You saw that with the new Help module there is now a "Help" tab in the General preferences, right? What would you say is missing there?

Yes, there are preferences for this module but what I mean is that there's no way to use something like "What's This?" in the Preferences to quickly get to the wiki page describing the selected preference.

yorikvanhavre, 2024-06-24:
Aah I see. The main problem here is probably that such wiki pages don't exist.
Would it make sense to have them? There might be a lot of work needed to maintain all that updated. Maybe it's wiser to invest in having better tooltips instead?

FEA-eng, 2024-06-24:
> Aah I see. The main problem here is probably that such wiki pages don't exist. Would it make sense to have them? There might be a lot of work needed to maintain all that updated.

There is this page describing many general preferences: https://wiki.freecad.org/Preferences_Editor and then pages for some workbenches: https://wiki.freecad.org/FEM_Preferences or https://wiki.freecad.org/PartDesign_Preferences for example. The "What's This?" tool could lead to proper sections of those pages. Wasn't the non-functional Help button supposed to do that?

yorikvanhavre, 2024-06-25:
> Wasn't the non-functional Help button supposed to do that?

TBH I don't remember :)

The problem is that those pages can quickly get obsolete... And not every WB has them