FreeCAD/FreeCAD #14847: No highlighting of multiple selections in Curves WB

EarlJay1, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I noticed how the behavior of multiple selections using the Crtl-Key changed. Somehow only the first element gets selected and highlighted, but the next ones I click don't, although they seem to be selected. 
It happens in the curves WB, where you often have to select multiple vertices, edges or planes.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.39
  * fasteners 0.5.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-22:
While the regression may well occur using the Curves Wb, it's not actually specific to that use case.

@bgbsww unfortunately there has been a further regression caused by https://github.com/FreeCAD/FreeCAD/commit/8e4c96d23bda93d2f6f5f21eaac6436306e7b1d9 , a really good test file and explanation can be found at: https://forum.freecad.org/viewtopic.php?t=88535

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37842 (Git)
Build type: Release
Branch: main
Hash: 83a268881dfce8473558a99ce0424a58f66f4893
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * SketchArch
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1

```

bgbsww, 2024-06-22:
Right you are!  Duplicated, fixed, patch coming.  I appreciate the excellent test.

What happened here was:
1) Narrowing down the second selection bug to failing to remove the TNP path from selection names and adding code that does that.
2) Bug report that it broke axis selection because the call that removes the TNP path should not have been called on static additions to the display like axis.
3) This bug report that it broke cases without a TNP path but with an elongated name ( Element inside Body inside Part ), which clarified that we need to explicitly test for a TNP path before applying the resolveElement results.  That's what will be in the PR ...

PR #14854 (2024-06-22): Toponaming: Check for selection mapped element name
Fix #14847.  Add (adjusted) proposed test from #13248.  Correct test exposed by #14801 .
closed 2024-06-22 by chennes