FreeCAD/FreeCAD #14848: 0.22Dev sketcher selection time 30 times slower than 0.21.2 for 100 edges on a high end PC

PhiJu5, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In sketcher, sketch SS96* (there is only one) select everything with mouse.
It's instantaneous with 0.21.2.
Do the same with latest 0.22dev: the selection takes a long time (W11 64b PC lot of memory, 5.6 Ghz, SSds, etc) : around 30 seconds (!!!!).

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * A2plus 0.4.65
  * AirPlaneDesign 0.4.0
  * Assembly4 0.50.13
  * CurvedShapes 1.0.9
  * Curves 0.6.39
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenDark 2024.3.13
  * Plot 2022.4.17
```

### Subproject(s) affected?

Sketcher

### Anything else?

As it's dev version, not expecting any fix. It's just to let 0.22 developpers aware of this.

[image]
[image]
[PerfPbm.zip](https://github.com/user-attachments/files/15936061/PerfPbm.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PhiJu5, 2024-06-22:
[image]
30 seconds later:
[image]

Syres916, 2024-06-22:
This is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/14618 which has been fixed since the build you are using was released. A new weekly build was released a few hours ago which is approx 140 commits newer than yours.

PhiJu5, 2024-06-22:
Opps sorry. Just updated to 'real' latest', and yes, fixed. Thanks.

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)

closed 2024-06-22 by PhiJu5