FreeCAD/FreeCAD #14849: Default solid appearance is hardcoded in files

pierreporte, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The default solid appearance is defined at theme level. It is saved in files, so that when they are opened in a FreeCAD with a different theme, the color is the same as it was when the solid was created.

Below is a picture illustrating the problem. I created a part with the classic theme, then opened in in the new default themes. They remained light grey, while parts created in the new theme are darker.

[image]

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.18
```

### Subproject(s) affected?

Core

### Anything else?

Similar to #13980

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-23:
I guess that is supposed to be like that, because the appearance is based on the material so you would loose the material if it's not saved in the file but any theme applies a material it wants. That's why I'm no fan of the default representation preferences and prefer the use of a default material.
@davesrocketshop can you confirm?

davesrocketshop, 2024-06-23:
The default material can be set in the preferences. Any theme can change it by providing it's own material file and setting that as the preferred default.

maxwxyz, 2024-06-23:
Yes but it is intended that the applied material on objects is stored in the file, right?

davesrocketshop, 2024-06-23:
Ok, I'm starting to understand the question.

Yes, when an object is created it is assigned the default material, which has the default appearance. Changing the theme won't reassign a new default material.

maxwxyz, 2024-06-23:
I'm closing this issue then as this is the intended behavior.

closed 2024-06-23 by maxwxyz

pierreporte, 2024-06-23:
Then why are themes allowed to change the default material? It’s confusing.