FreeCAD/FreeCAD #14853: A fillet or chamfer should support e.g. 2.00mm radius if the geometry is equal 2mm

Panda631, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem was discussed in an older thread from 2018. So I learnd that a minimal face must remain when filleting two faces.

[image]

But maybe the user could supported by the UI when put in an even value (2.0mm in the example)
1st:  Show 2,0mm radius in the input box, but calculate internal with the nearest possible radius (1,99999mm)
2nd: Replace the 2,0mm input automatical to the nearest possible radius.

I would prefer the first option.
Just an idea, what do you think?

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.21
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.1.23
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-06-22:
This topic is a little similar to mine, which I created earlier. 
There you can see opinions on this matter in the comments.
https://github.com/FreeCAD/FreeCAD/issues/13942

I assume that similar topics will appear many more times. Because this is indeed a problem that we often encounter.

If I understand correctly, then doing the 1st option is difficult, because it involves calculations.
But what surprises me is that comparing two numbers seems to be not a difficult task. And after comparison, you can, for example, automatically assign a difference of, for example, 0.1% of the specified values. This would be sufficient accuracy.

maxwxyz, 2024-06-23:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/5561

closed 2024-06-23 by maxwxyz