FreeCAD/FreeCAD #14858: Part Chamfer breaks when re-assigned

Jookia, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you create a Chamfer on a Chamfer of a Box in the Part workbench then set the outer Chamfer's base to the Box the chamfer will appear in the wrong place.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37842 (Git)
Build type: Release
Branch: jookia_testing
Hash: 0f9dcefb3f8a99969117ff60167e6a8fdb26706d
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.67

(Note: Branch base is f294a5efc546eda9f0d6d1b85603544f09053cbe)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-23:
Please add a screenshot to describe how to reproduce

Jookia, 2024-06-23:
Sure, it looks like this:

[image]

maxwxyz, 2024-06-24:
I'm not sure how to reproduce this and when it breaks, could you add more details and steps how to reproduce? TIA.
@bgbsww FYI

Jookia, 2024-06-24:
1. Open FreeCAD
2. Open the part workbench
3. Add a box
4. Click on edge of the box
5. Chamfer it
6. Click another edge of the box
7. Chamfer it
8. Select the second chamfer (Chamfer001)
9. Change its base to the box
10. It has the error 'Chamfer001: NCollection_IndexedMap::FindKey'

In  my case the EdgeLinks for the Chamfer001is "(<Part::PartFeature>, ['Edge15'])" where the part is the box. The element map does not use property tags, instead using 'Edge16', etc

Jookia, 2024-06-30:
I'm not sure if this is a TNP issue at this point. I believe this is the same issue as #14643 where deleting things in general breaks things.

luzpaz, 2025-07-07:
Thanks @zayats1. Not sure if it's necessary to play with the image sizes, btw. It's harder to discern the issue because of image dimension warping.

closed 2025-07-27 by maxwxyz